NVIDIA-AI-IOT/jetbot #9: Suggestion for 3D Printing Wiki

jwiki95, 2019-03-19:
The 3D printing Wiki only currently lists some printers, sample material, and manufacturers. Because the sample chassis is fairly complex, I think it would be beneficial to add some information on how to set up the print. There's a lot of good things that can be added to make sure prints are successful like

- Print orientation

- Support structure

- Infill density

- Wall thickness

Bondifrench, 2019-03-20:
Maybe also add model units? Are the stl in mm? inches? cm? My local 3D printing studio is asking.

jaybdub, 2019-03-21:
Great points.  Let me check with Chitoku.

If you guys try a print configuration that works well for you, please let us know.  I think collecting this info will be very valuable to others.

tokk-nv, 2019-03-21:
HI jwiki95, Bondifrench,
Both good points.
I added a little section under the 3D printing wiki page.
Please have a check.
https://github.com/NVIDIA-AI-IOT/jetbot/wiki/3D-printing

panzerfrank, 2019-03-22:
My printer tells me the chassis is too big to fit.  Could you tell us the chassis gross dimensions please

tokk-nv, 2019-03-22:
Hi panzerfrank,
The dimension of the chassis is following;

- Length; **136mm**
- Width; **90mm**
- Height; **54mm**

We designed the chassis so that it will fit the build volume of most of the 3D printers.

panzerfrank, 2019-03-23:
Thank you so much

acopylove, 2019-04-04:
Hi, @tokk-nv, could you describe the source files for 3d printing in details? 

I see the STEL, IGES and STEPS files – are they all defined the one models but in different formats or I can use only STL models? 

Why are there the 60 and 65 versions of the base and shroud? 

Which one should I use?

acopylove, 2019-04-04:
@tokk-nv I found answers on my questions on the Bill Materials page.

tokk-nv, 2019-04-04:
Hi @acopylove, 
Great that you found the answers in the Bill of Materials wiki page.

The STEP and IGES files are for the same model, but in different formats.
And you can pick 60mm configuration or 65mm configuration based on the wheels you pick.

closed 2022-07-15 by jaybdub