NVIDIA-AI-IOT/jetbot #10: Battery Life

geoffhall2013, 2019-03-20:
Can anyone comment on the battery life for jetbot? For example, using the recommended battery, how long can jetbot follow an object?
It would be a nice option to have the chassis hold two batteries. One for the Nano and one for the motors.

jaybdub, 2019-03-21:
Hey Geoff,

Haven't had time to do a full benchmark yet, but to give some rough idea; we run the collision avoidance sample for about 3-4hrs on a single battery charge.

That includes constantly running image resizing, WiFi streaming, resnet18 execution, and motor control.

Lasts much longer when not streaming / running full DL workloads.

Hope that helps some.

John

geoffhall2013, 2019-03-21:
That's a long time on one charge. Very good. Battery capacity does not seem to be an issue.

closed 2019-03-21 by geoffhall2013