NVIDIA-AI-IOT/jetbot #11: TF or Keras version

ShettyHarapanahalli, 2019-03-21:
Do we have any plans to release a keras or tf version for this ?

jaybdub, 2019-03-21:
Hi,

We have ROS port for motor control, etc:

https://github.com/dusty-nv/jetbot_ros

No plans on using one framework vs. another at this time, however, our object following example uses models exported from the [TensorFlow object detection API](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/detection_model_zoo.md) to TensorRT.   It works with SSD Mobilenet v1 / v2 for real-time object detection on Jetson Nano.

Best,
John

ShettyHarapanahalli, 2019-03-22:
Hi John,

Thinking of adapting this to my bot on TX2, can you please share more information about collision avoidance. I was thinking of point cloud date from a stereo vision for collision avoidance but this can be good. 

BR, 
SSH

jaybdub, 2019-03-24:
Check out the [examples](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Examples) page in wiki.

The collision avoidance sample trains an RGB image classifier to predict whether the robot is 'blocked' or 'free' and the appropriate action is then taken accordingly.   

If you're using a different platform (with different sensor placement), you would need to collect data for that platform.  If you use a regular camera, you could probably just modify the example to collect data and train a new model for your robot. 

John

closed 2019-03-25 by ShettyHarapanahalli