NVIDIA-AI-IOT/jetbot #12: JetBot 零件清单中国区购买链接汇总

unbug, 2019-03-23:
This issue is meant to help Chinese devs to buy the materials of JetBot in China. I'll keep updating it, andy help is welcome. 

Its appreciate leaving the issue open for a while, thanks.

中国区购买 JetBot 的零件比较麻烦，所以整理了一个可以在国内购买的列表，欢迎一起更新汇总。

https://github.com/unbug/NVIDIA-JetBot-Diy/wiki

jaybdub, 2019-03-25:
Thanks unbug, we look forward to seeing how your fork progresses.

John

unbug, 2019-03-26:
Hey @jaybdub, the 3D printer is hard to find, do we have a cardboard option?

jaybdub, 2019-03-27:
Currently we just have the 3D printed chassis design.

You could possibly build an alternate chassis however you like.  To work well, the most important aspect is to place the camera in the correct orientation.

If you design an alternative chassis in your fork, let us know!  Potentially this could be useful to others.

John

unbug, 2019-03-27:
Hum...I believe we can just buy a toy car model, so we can focus on AI developing base on Jetson. I'll see what can I find.

unbug, 2019-03-27:
Hey @jaybdub, I found some cardboard, the size is matched Jetson nano. What do you think?

### Model 1, Smart Motor Robot Car Chassis Kit

[image]

[Taobao](https://item.taobao.com/item.htm?spm=a230r.1.14.160.4e3058d58fdH3n&id=562562707881&ns=1&abbucket=1#detail)

[image]

[image]

[Amazon](https://www.amazon.com/Emgreat%C2%AE-Chassis-Encoder-wheels-Battery/dp/B00GLO5SMY/ref=sr_1_22?crid=3CQA2604A0IIZ&keywords=arduino+car&qid=1553685564&s=gateway&sprefix=arduino+car%2Caps%2C361&sr=8-22)

### Model 2,
[image]

[Taobao](https://item.taobao.com/item.htm?id=578766841591&ali_refid=a3_430008_1006:1121095167:N:ihrAaDEdgmEOJJfmUhZeDnYdl%2BSOiVUP:d6e2607318c201b66fc2b10bc5ccb42a&ali_trackid=1_d6e2607318c201b66fc2b10bc5ccb42a&spm=a230r.1.0.0)

### Model 3,
[image]

[Taobao](https://item.taobao.com/item.htm?spm=a230r.1.14.87.127c6d0fbNWE5g&id=566000074468&ns=1&abbucket=1#detail)

### Model 4,
[image]

[Taobao](https://item.taobao.com/item.htm?spm=a230r.1.14.96.4e3058d58fdH3n&id=589010397546&ns=1&abbucket=1#detail)

unbug, 2019-04-03:
Hi @jaybdub do you have an answer to my proposal?

jaybdub, 2019-04-03:
Hi unbug,

Cool!  It looks like model 2 is the closest match and would likely be compatible with the other electronics in the bill of materials.

It seems the kit only includes the motors, wheels, caster and chassis.  So you would need to buy all of the other components.

You would also need to figure out how to mount the camera and Jetson Nano, as well as how to house the battery.

Best,
John

unbug, 2019-04-04:
Thanks, @jaybdub, model 1 works the same as model 2, so I think both of them are matched.

jaybdub, 2019-04-05:
They look pretty similar, just a few minor differences:

It looks like (2) uses the same motors in our bill of materials, so will work directly with the motor driver we list.

Also, we noticed that the cheap continuous rotation servos (like those used in (1)) tend to 'drift', and require frequent calibration.  Something to keep in mind!

Best,
John

unbug, 2019-04-05:
Thanks, @jaybdub.
Indeed, the drift is an issue.

mayuanjason, 2019-04-12:
Hi unbug,

You can try mbot, a metal car, please refer to the following webpage for more detailed info.
https://www.makeblock.com/steam-kits/mbot

unbug, 2019-04-13:
@mayuanjason Looks cool, have you tried it? Is it fit sie of Jetson? I'll investigate further, thanks.

mayuanjason, 2019-04-13:
Hi unbug,
No, but I think this plan can work.
mBot has a Arduino board on it, jetson nano can communicate with Arduino through serial port, so the control path is something like below:
Command from jetson nano -> serial port -> Arduino -> motor(e.g. start, stop)
And we can use pyserial library for serial communication from jetson side, and there also has some serial related functions in Arduino side.

unbug, 2019-05-05:
Good news @jaybdub , check out some showcase of Jetbot build by Chinese devs, some of them even success use a third party robot car model. https://github.com/open-ai-robot/NVIDIA-JetBot-Diy/issues/6

wujiang, 2019-06-19:
We created a kit for anyone to order online https://curious.voyage/

unbug, 2019-06-19:
@wujiang thanks. Do we have to pay the shipping fee and customs duties if we order to China?

wujiang, 2019-06-19:
@unbug np. Yes, you'll need to pay for the shipping. Not sure about tariff.

jasper-hue, 2019-10-05:
Hey @unbug , 

I tried your approach of using PCA9685 as the motor driver, 
But can you explain to me how can we connect the dc motor in that. 
PCA9685 is a servo driver and has 3 pins Gnd , V+ and PWM, the DC motor only has two. 
Any idea how you are planning to connect them together ?  

Thanks in advance.

unbug, 2019-10-05:
Here are some success cases https://github.com/open-ai-robot/awesome-nvidia-jetson/issues/6

closed 2022-07-15 by jaybdub