NVIDIA-AI-IOT/jetbot #14: Can we buy the whole jetbot parts online?

AK51, 2019-03-25:
I have printed out the stl and tried to put the electronics on it before my nano arrive. However, there is no wheels and the screws holes do not really fit my electronics. I wonder if I can just buy the robot parts. Thanks.

jaybdub, 2019-03-25:
Hi AK51,

Thanks for reaching out!

Currently you will need to buy the components individually.  We understand that some components may come in and out of stock (especially since this project is so new).  Hopefully the community will develop forks of this project to support alternate hardware configurations!

That said, because this project is 100% open source, it is possible that distributors will create their own kits that are compatible with the JetBot software.

If you make a fork of this project using alternate hardware please let us know!  I think it will be very helpful to the community.

Thanks!
John

darraghdog, 2019-04-12:
At least for the 3d parts sellers seem to be appearing : https://www.tindie.com/products/benwynne/3d-printed-parts-for-nvidia-jetson-nano-jetbot/
Disclaimer: I have not bought from there so cannot vouch for it. (But do plan to)

wjzhou, 2019-05-09:
I will try to create a kit, is there any interest here?

Disclaimer: I'm not a distributor. I'm building the jetbot for my 7 years old daughter. After it was build, several of her classmates and friends want one too. Sourcing the individual components for their parents are overwhelming for me. So, I think it's time to create a kit.

If there is interest, I'll buy extra parts to reduce the individual cost.

wujiang, 2019-06-19:
We created a kit here in case anyone is interested: https://curious.voyage/

HannahZhangVW, 2019-06-19:
https://www.ebay.com/itm/383013555802
The above link sell a package of 11 parts. 
The rest items you need to buy ares:
1. Jetson naon developer kit, https://www.amazon.com/NVIDIA-Jetson-Nano-Developer-Kit/dp/B07PZHBDKT/
2. camera at https://leopardimaging.com/product/li-imx219-mipi-ff-nano/
3. battery and usb cables: 
4. Mircro SD card.
5. 3D printed set.

closed 2022-07-15 by jaybdub