NVIDIA-AI-IOT/jetbot #16: Size of power bank?

port513, 2019-03-27:
Since there is no way to ship to Sweden if it's LiPO, what size of power bank should I choose?

tokk-nv, 2019-03-29:
The battery bay of the chassis is following, so the battery needs to be smaller than this.

- Length: **136mm** 
- Width: **71mm**
- Height: **20mm**

I hope you can find a compatible battery.
Then you can make a fork of this project to show the Swedish version of the BOM.

agrobler58, 2019-04-03:
Same issue here. Suggested INIU model is hard to obtain in Australia. Regarding power, do both ports need to have 3A? Or is it OK if one of them (for motors) has 2.4A?

aussetg, 2019-04-06:
I don't think the battery is capable to output 5v 3a on each ports at once, it's probably 3a in total. If they say it can it's probably a lie tbh.

einsteinagogo, 2019-04-18:
This could be make or break for Europe, trying to find a battery which fits. I've just ordered the following..

which supports 3A/5V on both USB ports, but it's maybe too wide..

https://www.amazon.co.uk/gp/product/B07KMRCM6Y/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

We will see....

http://andysworld.org.uk/2019/04/18/can-you-guess-what-it-is-yet/

einsteinagogo, 2019-04-18:
where does the 3A/5v come from as the Description of that Power Bank states

Output: 2.4A Output for each 2 USB port

davinderc, 2019-05-17:
@einsteinagogo where you able to make it work? I ordered an Ansmann 10.8 QC 3.0, but then ran into a problem using the USB C output because of the QC 3.0 protocol.

einsteinagogo, 2019-05-17:
@davinderc I've been trying MANY different power banks available in the UK, I now have two which work, BUT they DO NOT FIT!!!

e.g. they correctly BOOT the NANO, rather than just booting and dying.

Does this Ansmann 10.8 QC 3.0 fit ?

I've now been discussing power banks for Import from China, which fit and deliver 3A/5V from both USB connectors.

I'm now ready to start building this weekend, after having a delay in obtaining original parts, and also printing the chassis which has been done.

davinderc, 2019-05-17:
@einsteinagogo The Ansmann 10.8 QC 3.0 fits, and is just slightly loose vertically. It boots the Nano just fine, although if I remember correctly, you need to push the button on the powerbank before plugging the nano in. But you will need to resolve the QC 3.0 issue. I borrowed this [adapter](https://www.digitec.ch/en/s1/product/dell-usb-c-to-usb-30-vga-hdmi-ethernet-lan-notebook-adapters-5668957) to be able to use the USB C output. 

By the way, as far as I know, standard USB (non-C) connectors are not built for more than 2 or 2.1 A, and most powerbanks of this size don't provide more than 2.1-2.4 A combined from both ports, so I would be wary of any powerbanks that state they can provide 3A without using a USB C or other port. That much power output seems very uncommon for this size powerbank. I'm going to measure the power consumption of both the board and the driver on Monday, so I can find a more suitable powerbank (without QC 3.0).

einsteinagogo, 2019-05-17:
@davinderc I just going by the Power Bank which is stated from the Bill of Materials, it states 3A from both ports...

I've test current consumption at BOOT and it does not seem alot, but clearly I have many power banks, which do not work, they start the device, but later go off.

davinderc, 2019-05-17:
FYI, the one you linked is wider and thicker than the dimensions cited at the top of this page.

Yeah, it says that, but that's advertising. If you try it, good luck and let us know how it works. I personally wouldn't buy it without knowing for sure and that's why Monday I intend to measure current first. Also, the current consumption will go up when you use the GPU for CNNs on images.

einsteinagogo, 2019-05-17:
yes I stated that "but it's maybe too wide.".

but it works fully, and I've got two different units which are working, they'll just fit in the battery compartment!

The other's are arriving direct from China, and are test samples, if they work I'll report here.

davinderc, 2019-05-21:
So I've found around 1.5 A peak power draw by the Nano while running obstacle avoidance (note I am running it at the half-power setting) and 1.5 A locked motor current draw if I set the motors to 0.9 and 0.75. You can run the whole thing on 2 A if you don't set the motors too high.

einsteinagogo, 2019-05-21:
Now I have my Jetbot built, I'm experimenting with a power bank which fits, and having no issues currently going through the examples, at half-power-setting-5watts.

(this power bank booted before and went off before, but silly me did have ethernet, usb mouse, usb keyboard and HDMI connected! - I suspect ethernet port is the issue!)

other than I think one of my TT motors is duff, because the balance is wrong, e.g. at 0.3 right motor does not drive....so new motors on the way to experiment more.

davinderc, 2019-05-22:
The motors are usually not the same, so I found that I had to simply set their velocities proportionally different. Even if you get new motors, you might find those are not the same also. For me 0.3 was the minimum to drive the motors, but a little more was better.

Also, I bought an external power supply for the Jetson so that I could run it without the powerbank if I wanted to. We used that to power it up and configure it to connect to the Wifi first, then afterwards just booted and waited for Wifi connect and commanded it through the Jupyter notebook. We had no issues with lack of power after that.

einsteinagogo, 2019-05-22:
@davinderc thanks for that tip about motors! that explains that!

Looks like I'll have to find the sweet spot for forward!!! then otherwise I'll go round in circles!

I already have power supplies which I use... but not good tethered...

davinderc, 2019-05-22:
Oh yes, the power supplies are just useful for initial configs or when you don't intend to run the motors. It's horrible to run the motors with it tethered. 

As for the motors, once you find a proportion that keeps the motors moving straight (took me 2-3 tries using the set_motors method), you can probably set those as constants somewhere in your code. I couldn't figure out how to do it when I was looking at it, but I did see that the source code has a way to do it so that you can simply call the forward method and forget about the underlying motor differences.

einsteinagogo, 2019-05-25:
The battery in the BOM is NOW available in the UK via Amazon after discussions with Topstar (INIU)

https://amzn.to/2VRAwWC

davinderc, 2019-05-25:
CC: @tokk-nv looks like we've got advertising using Amazon affiliate links...

ergoen, 2019-06-06:
> Since there is no way to ship to Sweden if it's LiPO, what size of power bank should I choose?

Sorry for writing in this old(-ish) topic, but where did you manage to order the correctly sized plastic ball in Sweden? I have been to various stores that I would imagine has items like this but have been met with odd stares.. :P

davinderc, 2019-06-06:
> Sorry for writing in this old(-ish) topic, but where did you manage to order the correctly sized plastic ball in Sweden? I have been to various stores that I would imagine has items like this but have been met with odd stares.. :P

So I found it easiest by searching for Pololu products. I took a look around and couldn't browse the Swedish shops all that easily (maybe because I'm in Germany or maybe because I don't understand Swedish). Not sure if this helps, but there's a shop in Finland that seems to have it: http://robomaa.fi/robotics/pololu-pallolaakeri-25cm-muovi

I ordered this from here in Germany and it works really well: https://www.exp-tech.de/zubehoer/sonstige/6715/pololu-ball-caster-with-1-plastic-ball-and-ball-bearings
This is Pololu's original page on it, maybe you can browse the Swedish distributors better than I can: https://www.pololu.com/product/2691

davinderc, 2019-06-06:
According to this Google image search on the site, they have it, but I can't seem to open their site, no idea why... Are you guys suffering from internet issues? (a traceroute seems to get to Swedish domains, and then times out)

https://www.google.com/search?q=site:www.lawicel-shop.se+st%C3%B6dkula&tbm=isch&source=univ&client=firefox-b-d&sa=X&ved=2ahUKEwjXx6aW_dTiAhV_ThUIHUKzCLYQ7Al6BAgAEAk&biw=1762&bih=886&dpr=1.09

davinderc, 2019-06-06:
EDIT: got it, here it is: https://www.lawicel-shop.se/pololu-ball-caster-with-25mm-plastic-ball
@ergoen

ergoen, 2019-06-06:
Wow thanks @davinderc ! It seems I was searching for the wrong term :)

davinderc, 2019-06-06:
It's kind of funny, I was searching for the wrong term here in Germany as well and then I asked a co-worker and a couple of searches and minutes later he had found exactly what I needed! By the way, I just used the ball and printed the caster they provided, works great!

ergoen, 2019-06-06:
Good to know! I have ordered one as well now! Thanks again

ChristophWenk, 2019-10-23:
The power bank used in the BOM is not available here in Switzerland.

So I have tried the goobay Quick Charge Qualcomm (10000mAh) but it just booted the Nano for like 30 seconds and then it shut down again. Sometimes it was able to boot the Nano but it shutdown immediately when I tried to use the motors. 
It only worked when I powered just the Nano with it and the motors separately with my computer. But that cannot be a solution.

Which alternative power bank that fits is able to support the power draw from all the components?

davinderc13, 2019-10-23:
@ChristophWenk I believe I mentioned one option that worked for me up above.

ergoen, 2019-10-23:
I bought a "GP Powerbank Voyage 2 / 10000 mAh", this works and fits also

ChristophWenk, 2019-10-23:
@ergoen Thanks for the suggestion. The specifications mention that it has 2 x USB-A DC 5V/2.4A. Does that mean the 3A for each port mentioned above are not needed after all?

ergoen, 2019-10-23:
I suspect it's the total combined power that has to support 3A, not per port. So any Powerbank that has a 3A usb-c port should in theory be fine (?). The goobay one you linked should therefore also work.. The specs seem identical to the GP one I have.

ChristophWenk, 2019-10-23:
Aha. But then you also used a USB-C cable instead of the USB-A mentioned in the BOM? Sorry for all the questions, I am not particularly familiar with this kind of stuff.

ergoen, 2019-10-23:
I'm only using the two USB-A ports, not the C. I didn't really do any research before buying it, just that it could output 3A :p

ChristophWenk, 2019-10-23:
Ok. The same as I did then. I just ordered a unit of the model you mentioned and will try get it to work with that one.
I will post an update if I was successful.

davinderc13, 2019-10-23:
@ergoen @ChristophWenk it would be wonderful if you guys could chat somewhere else about this, as I don't enjoy receiving notifications about the same discussion I had long ago (and is clearly described up above, although Christoph doesn't feel like reading it). I've tried unsubscribing from this thread and I'm still receiving emails with your discussions. Please, could you discuss it elsewhere?

davinderc13, 2019-10-23:
@tokk-nv could this issue be closed, since it has already been thoroughly discussed and resolved?

ChristophWenk, 2019-10-23:
@davinderc13 Just to be clear: I read through the whole issue. All I found was workaround solutions. I'd like to have a clean solution that works out of the box like the one mentioned in the BOM.
Also, there is an notifications "Unsubscribe" Button on the top right.

davinderc13, 2019-10-23:
@ChristophWenk then you've misunderstood what I wrote or intentionally ignored it. I had the exact same questions that you did and discussed/resolved them.

As for clicking the unsubscribe button, that's a "workaround" as you put it, because I did click it and find that I am once again subscribed to it. Please open another issue rather than dig up an old one and pester other people.

davinderc13, 2019-10-23:
@tokk-nv as I mentioned before, the unsubscribe button does not seem to have any effect and the original OP has not followed up on this thread, nor does it seem there will ever be a resolution that will satisfy everyone around the world. Could you please close it so I can stop receiving emails about it?

closed 2022-07-15 by jaybdub