NVIDIA-AI-IOT/jetbot #17: Is the 3D model open source?

wujupi, 2019-03-28:
A 60 mm diameter TT motor tire was not found on the Chinese e-commerce website. This is not a common commodity. The 65mm diameter TT motor tires are very numerous and have been widely used in various intelligent robot cars. So I want to modify the 3D model to fit the 65mm wheel.
thanks.

wahid, 2019-03-29:
Wouldn't the 65mm tire work without motification?

sirwalterrrich, 2019-03-29:
@Wujupi You can convert the .stl file to a solid using software like fusion360 and then modify it to your liking.  How to video link below.

https://youtu.be/yO6hboYWnfM

tokk-nv, 2019-03-30:
Hi wujupi, and all,
We also realized 65mm wheels are more widely available, so we are trying to accommodate that.
If we just swap the 60mm wheels with 65mm wheels, it would tilt the robot chassis and change the camera orientation, so we'd want to also raise the caster part by 2.5mm.

For now, I designed the caster parts designed for 65mm wheels and put them (along with STEP and IGEL files) under a branch.
I'm making a pull request, so for now please check the branch.
https://github.com/NVIDIA-AI-IOT/jetbot/pull/20

wujupi, 2019-04-03:
@sirwalterrrich @tokk-nv Thank you, Thank you, I have printed the 65mm model and installed it on the jetbot, it works perfectly.

storagejarett, 2019-04-03:
The body is really poorly designed for modularity.  If a change to wheels needs 100% remodeling and reprint, it won't work for most people.

This should be a modular design, where they let item is at a fixed location/height and the rest of the components can be modified individually. 

There is way too much design around Adafruit/Amazon specific parts which are not guaranteed availability in all countries. 

Also, the existing design, no matter what the orientation, has massive amount of overhanging elements which can be difficult to print without massive amount of supports. 

Key points:. Height and position of camera. 

In my build, I am:
* using 2 smaller individual motor drivers instead of the long double driver
* Using either 18650 batteries or a 4s lipo pack + buck power converter to bring 12-14v down to 5v
* Using a different camera
* Possibly using tracks instead of wheels + caster

There should be camera location mount with optional camera mounts.  A base body it connects to. And the rest of the components have modular attachment. 

The Nano board on top may not be the best way to go about this either. If it was bottom mounted or in the chassis, all the other parts could build around and on top of it.

Agent-Cain, 2019-04-23:
What is supposed to be mounted in the lower front ?

closed 2022-07-15 by jaybdub