NVIDIA-AI-IOT/jetbot #18: Adafruit MotorHat

maximecb, 2019-03-28:
Hi. I see that the Jetbot is using the Adafruit MotorHat library, which was originally designed for the Raspberri Pi. I'm currently [running into issues](https://devtalk.nvidia.com/default/topic/1049116/jetson-nano/python-gpio-support-/) trying to get the RPi.GPIO library to work. I was wondering if you had any pointers on how to get the Python RPi GPIO libs to work on the Jetson Nano? Are there packages to install, etc?

jaybdub, 2019-03-28:
Hi maximecb,

The Jetson.GPIO library (which is actually also importable as RPi.GPIO) is already pre-installed if you're using Jetson Nano or the latest JetPack.  It's located at  ``/opt/nvidia/jetson-gpio``.  

Make sure that you uninstall other versions of RPi.GPIO that you may have installed.

Then, there's a bit of extra setup that needs to be done to use the Jetson.GPIO library

#### Step 1 - Setup permissions

Execute the following from a terminal on your Jetson

```bash
sudo groupadd -f -r gpio
sudo usermod -a -G gpio $USER
sudo cp /opt/nvidia/jetson-gpio/etc/99-gpio.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && sudo udevadm trigger
```

Then, reboot the Jetson

```bash
sudo reboot now
```

#### Step 2 - Extend python path

Add the following in your python script before calls to import the RPi.GPIO library (or libraries that use it like the motorhat library)

```python
import sys
sys.path.append('/opt/nvidia/jetson-gpio/lib/python/')
sys.path.append('/opt/nvidia/jetson-gpio/lib/python/Jetson/GPIO')
```

You should then be able to import the Jetson.GPIO 

```python
import Jetson.GPIO
```

or equivalently,

```python
import RPi.GPIO
```

I haven't tested this with the MotorHat library yet, but I was able to import RPi.GPIO after these steps.

Let me know if you run into issues, or if these steps work for you.  I think this will be valuable to others.

Best,
John

maximecb, 2019-03-28:
Thanks for taking the time to write these instructions John. I am able to import the lib and make an LED flash. I will look again at Pypot (I2C) and the motor shield in the next couple of days.

maximecb, 2019-03-28:
I2C seems to work as well. Thanks again John!

jaybdub, 2019-03-28:
Awesome!  Thanks for verifying, this will be helpful for others.

Let me know if you run into other issues.

John

maximecb, 2019-03-29:
Small update: I've been made aware that the Jetson.GPIO package is now available through pip:
https://pypi.org/project/Jetson.GPIO/

Also, there doesn't seem to be PWM support available on the GPIO package that comes with the Jetson OS image. For my application (driving a DC motor), software PWM is probably doable, but hardware PWM would be even better.

jaybdub, 2019-03-29:
You're correct.  Jetson.GPIO doesn't support PWM.  However, the following I2C controlled board can generate PWM signals up to ~1.6kHz and should have Python libraries available. 

https://www.adafruit.com/product/815

You could potentially connect this to whatever H-Bridge meets the power requirements of your application.  The motor driver we used for JetBot actually combines this chip and multiple H-bridge on the same board.

https://www.adafruit.com/product/2927

Depending on what your application, that may just work!

John

weiqing-beijing, 2019-04-06:
Hi @jaybdub , thanks for the guide. Very helpful!

One more question regarding the spi and i2s. The document mentioned there are spi and i2s pins. Do you know how to activate and use those pins? Or if there are any documentation as reference. 

Thanks, wq

closed 2022-07-15 by jaybdub