NVIDIA-AI-IOT/jetbot #21: image size over 64GB of the jetbot image

weiqing-beijing, 2019-04-01:
When I tried to write the jetbot image file to my 64GB microSD card, the downlaoded image zip file size is 6.8GB but the actual image size after unzip(by Etcher SD Flash Utility(v1.4.9) or opened with WinRAR (v5.6 64-bit)) is 64.02GB which could not fit into a 64GB microSD card. Before I go for a 128GB microSD, the question is if the image file is indeed a 64GB size, because from the instruction it shows 16GB is minimum requirement for microSD card. If it was 128GB at minimum, the instruction need to be updated.

[image]

wujupi, 2019-04-03:
I also have the same problem, the image must use 128G TF card to burn。

weiqing-beijing, 2019-04-03:
Seem it is. I have tried to make the image manually by following the [Create SD Card Image From Scratch](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Create-SD-Card-Image-From-Scratch) instruction under Jetbot wiki, it works and it only took up 20G using ```df``` command. So I just rebuilt a new SD image from scratch on a 32G micro-sd card.

jaybdub, 2019-04-04:
Hi, 

Glad to hear you found a suitable workaround.

I'm re-opening this issue to help people find it, as well as track a potentially more convenient solution.

Thanks,
John

weiqing-beijing, 2019-04-06:
Thanks, John!

By the way, I have got hold of a 128GB micro-sd card and successfully flask the image. As @wujupi mentioned, the image should be based on a 128GB system. I am working on both 64GB(self made per guidance) and 128GB(flashed based on the official jetbot image)  with no problem. Though I do think nVidia should offer a minimum-sized image for people to decide by themselves if they needed to work on a bigger sd card.

sundar19, 2019-04-19:
@weiqing-beijing  I tried to create SD card image from scratch and followed all the instructions as specified in https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Create-SD-Card-Image-From-Scratch but when I try to access using nano's  IP address from my laptop , jupyter notebook doesn't open and says page not found pls help

weiqing-beijing, 2019-04-23:
@sundar19 I noticed that the jupyter notebook service did not load as a boot service(with an error message saying the service could not be loaded during boot procedure). I have not got time to debug it so just ssh into it from host pc with `ssh USER_NAME@JETSON_NANO_IP_ADDR`. Once logged in,  enter `~$ jupyter notebook` to start the jupyter web session.

sundar19, 2019-04-26:
I renamed my nano as jetbot and it worked out, is it compulsory to name it as jetbot?

idcrook, 2019-05-05:
I encountered as issue when imaging `jetbot_image_v0p3p0_63GB.zip` on macOS with Etcher.app onto 64GB Sandisk µSD card: Jetson Nano would boot into recovery mode and not boot the OS. I thought µSD card was problem, but suspected µSD worked as expected when I used **sdkmanager** to image Jetson OS.
https://devtalk.nvidia.com/default/topic/1051276/jetson-nano/new-jetson-nano-not-booting-instead-showing-quot-recovery-quot-screen/

romilly, 2019-05-10:
@jaybdub  I'd be happy to post/host a prepared 32 MB jetbot image to save others the installation time, but I am concerned about IP issues as I don't work for NVIDIA. What's your advice?

ManuelZ, 2019-06-09:
Do you have the SHA sums of the two images? I'm getting 12 partitions in my USB flashing it with Etcher using the 63GB, which seem a bit too many.

romilly, 2019-06-09:
12 - That sounds right.

ManuelZ, 2019-06-09:
Really? Thank you!

tokk-nv, 2019-07-08:
Hi all,
We now have an expandable SD card image for JetBot.
It is linked from the [Software Setup wiki page](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup). Please have a check.

This image should work on any SD card greater than or equal to 32GB, and expands the rootfs/APP partition to use up all the SD card space on the first boot in the same way the official SD card image does.

fspiga, 2019-10-01:
@tokk-nv, for some reasons the following links do not seems to work:
* https://drive.google.com/open?id=1GF2D814hkViwluZ5SgNKW56cQu_5Ekt5 (jetbot_image_v0p3p2.zip)
* https://drive.google.com/open?id=154LYkrFiBHFVV2kLTjRSwdV_vbZNdpxH (archived 63GB)

closed 2022-07-15 by jaybdub