NVIDIA-AI-IOT/jetbot #23: Is this project compatible with Jetson TX1 since it is also based on Maxwell arch?

panpengjun, 2019-04-03:
Thanks!

jaybdub, 2019-04-03:
Hi panpengjun,

The robot chassis itself is designed to fit the Jetson Nano, but many of the software components would still work.

Are you looking to use a particular piece of software from the project?

Best,
John

panpengjun, 2019-04-03:
Hi John,

I am working on a project based on TX1, which involves a little bit of everything, including sonar sensors, motor, camera, etc. It uses ROS, which makes the code base relatively larger than your Nano project. I am wondering what needs to be done to duplicate this project on a TX1 since I don't have a Nano yet, just as a test drive.

PJ

rangeli, 2019-04-06:
You would need a carrier board like the [Auvidea J90-LC](https://auvidea.eu/product/70761/).  You would also need to modify the chassis to fit that carrier board.  I'm not sure if the I/O on that board would be sufficient.  The power requirements will probably differ as well.  The other annoyance with the TX1 is storage: it has 16GB eMMC and no built-in card slot which may prove limiting.  It would be nice to see a fork of this project for the TX1, though.

closed 2022-07-15 by jaybdub