NVIDIA-AI-IOT/jetbot #25: Code Invalid in Wiki

gizmohd, 2019-04-05:
The code given in the Wiki is invalid and doesnt work. DLI_Jet_Demo

jaybdub, 2019-04-05:
Hi Gizmohd,

Thanks for raising this issue.  Apologies, I believe the promotion code has expired.  We will update the Wiki to reflect this.

Instead, you can simply download and follow the [training notebook](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/notebooks/collision_avoidance/train_model.ipynb) on a system with PyTorch installed.  We've actually trained a collision model on Jetson Nano itself, which may be a fun starting point!

If you follow the step "install latest software" in the software setup, it should include the training notebook.  Be sure to download / backup any data you have collected before hand, just in case.

Please let me know if you run into any other issues :)

Thanks,
John

gizmohd, 2019-04-05:
That’s crazy... device has been out a week and the code is expired?  ;)

closed 2022-07-15 by jaybdub