NVIDIA-AI-IOT/jetbot #27: Caster Ball Supplier in Europe or other options?

davinderc, 2019-04-07:
I've tried to find the same caster ball here in Europe, but could not find the exact same thing. Does anyone have good options for this? Could I use the following caster and would it be a lot of work to modify the chassis STL for this?
https://www.digikey.de/product-detail/en/sparkfun-electronics/ROB-08908/ROB-08908-ND/5814243

weiqing-beijing, 2019-04-07:
I just 3D printed into PLA by myself. It works well. Just polished the surface a little bit. Also you might be aware that there is a new 65mm caster shroud file in asset folder as there are more choices of 65mm wheels for tt motor than the 60mm ones. You can also 3F print your wheels if you want.

I developed it in OpenSCAD. Here is the [STL output](https://github.com/weiqing-beijing/jetson_projects/blob/master/3D_Design/jetbot_caster_ball%20_shaft_adapter.stl). You can download the original OpenSCAD [file](https://github.com/weiqing-beijing/jetson_projects/blob/master/3D_Design/jetbot_caster_ball%20_shaft_adapter.scad) to customize it per your needs.

Enjoy!

wq

davinderc, 2019-04-07:
The thing is, I'm doing this for my company and the 3D printer is at an office in a different city, so I don't really have the resources or time to try printing it out to see if I can make it work. I've already had to adapt other components, so our hope is that we can simply order the right parts and get it working out of the box.

aussetg, 2019-04-07:
[Bought this one, we'll see](https://www.aliexpress.com/item/25-4mm-1-inch-1pcs-Delrin-POM-Plastic-Solid-Balls-for-Valve-components-bearings-gas-water/32864674491.html?spm=2114.search0604.3.2.54466d1eDUqMLp&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10068_319_10892_317_10696_10084_453_454_10083_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103,searchweb201603_61,ppcSwitch_0&algo_expid=5c01b60a-6d73-4470-9230-534c3b8d6760-0&algo_pvid=5c01b60a-6d73-4470-9230-534c3b8d6760&transAbTest=ae803_5)

davinderc, 2019-04-07:
This looks interesting aussetg, although it seems to be from China, so I wonder how long it will take to ship to Germany?

aussetg, 2019-04-07:
In my experience it will take between 2 and 4 weeks.

There seems to be UK sellers too

https://www.ebay.fr/itm/25mm-Diameter-Solid-Delrin-Polyoxymethylene-POM-Celcon-Plastic-Balls-/142124480992

chitoku, 2019-04-11:
The same size ball should work as long as it has low friction.

Some keywords are;
- "Derlin" (product name)
- "Duracon" (product name)
- "POM"
- "Acetal Co-polymer"
- "engineering plastic"
- "low friction plastic"

I hope we can find something equivalent in different regions.

davinderc, 2019-04-11:
I found one for Germany! For some reason my initial search wasn't giving any interesting results, but a co-worker founds this, which might even be adaptable to be used without the 3D-printed shroud, but would likely need adaptations of the chassis.

https://www.exp-tech.de/zubehoer/sonstige/6715/pololu-ball-caster-with-1-plastic-ball-and-ball-bearings

Agent-Cain, 2019-04-11:
> I found one for Germany! For some reason my initial search wasn't giving any interesting results, but a co-worker founds this, which might even be adaptable to be used without the 3D-printed shroud, [...]

I've the same problem that i can't find one in germany. Did you order that ball caster ?

davinderc, 2019-04-11:
> I've the same problem that i can't find one in germany. Did you order that ball caster ?

Yep! Even if the hardware doesn't work, the ball is 1 inch, so it should do just fine. In any case, I'll update on how it works once it arrives and the chassis is printed.

Agent-Cain, 2019-04-13:
> Yep! Even if the hardware doesn't work, the ball is 1 inch, so it should do just fine. In any case, I'll update on how it works once it arrives and the chassis is printed.

Where from do you order your wheels ?

davinderc, 2019-04-13:
> Where from do you order your wheels ?
You can order them here as well, although they are not the same wheels. If you search exp-tech for wheels, there are some thinner wheels, but I'm not sure if they fit on the motors properly.
https://www.exp-tech.de/plattformen/robotik/sonstige/6536/wheel-65mm-rubber-tire-pair

aussetg, 2019-04-13:
So now we know we can just check Pololu retailers :)

Which is nice, I'm going to use the opportunity to buy step down converters to power by battery the jetbot in 10w mode

einsteinagogo, 2019-04-18:
> I've tried to find the same caster ball here in Europe, but could not find the exact same thing. Does anyone have good options for this? Could I use the following caster and would it be a lot of [...]

I've just taken delivery of the caster ball from... The Bay!

1 inch Diameter Solid Delrin Polyoxymethylene (POM) / Celcon Plastic Balls

https://vod.ebay.co.uk/vod/FetchOrderDetails?
itemid=131946220673&transid=1577497806003&ul_noapp=true

parts arriving here for Jetbot

http://andysworld.org.uk/2019/04/18/can-you-guess-what-it-is-yet/

nullr0ute, 2019-04-28:
I found this in the UK, the term bearing brings up different types of bearing so it was hard to find the direct equivalent until I worked out better search terms. 

https://www.ebay.co.uk/itm/Single-Delrin-Plastic-Ball-Metric-Sizes-1mm-to-50mm/372119091645

davinderc, 2019-05-17:
I've put it all together and used the Delrin ball from Exp-Tech. It worked wonderfully!

https://www.exp-tech.de/zubehoer/sonstige/6715/pololu-ball-caster-with-1-plastic-ball-and-ball-bearings

closed 2019-05-17 by davinderc