NVIDIA-AI-IOT/jetbot #30: Timeline of the availability of spi and i2s function on board

weiqing-beijing, 2019-04-09:
Hi, has anyone down here got information of if and when the spi and i2s would be available to program on Jetson Nano?

I was able to use Jetson.GPIO and Adafruit Blinka to work on GPIOs and two [I2C buses](https://github.com/weiqing-beijing/jetson_projects/tree/master/testing) on Jetson Nano. It is very convenient. I am planning to tap into the programming of spi and i2s but realized the lib was not ready yet. And I think uart is not ready either, though I haven't tested yet.

Thanks!

kunguz, 2019-04-11:
I don't have an answer for you, and not in position to answer it as well. But just saying, `i2cdetect` works with a Jetson Nano. I am not sure but may be it is already there since such a command works in shell of a Jetson Nano's Ubuntu OS:

```
i2cdetect -y -r 0
```

weiqing-beijing, 2019-04-14:
Thanks @kunguz. I have posted a request on [Jetson nano Developer Forum](https://devtalk.nvidia.com/default/board/371/jetson-nano/1/) and so far no firm answer yet. The i2c indeed works perfectly. I am using both i2c ports now. Very convenient.

kunguz, 2019-04-14:
@weiqing-beijing for SPI, I was told to use this link for TX1. This may be helpful for you as well: [Jetson Nano Developer Forum link](https://devtalk.nvidia.com/default/topic/1049916/jetson-nano/driving-an-adafruit-2-8-with-a-jetson-nano/post/5329393/#5329393).

closed 2022-07-15 by jaybdub