NVIDIA-AI-IOT/jetbot #34: What is the maximum power draw of the Nano ?

aussetg, 2019-04-14:
Is 15w enough ? Or should you prepare for 20w ?

tokk-nv, 2019-04-18:
Hello aussetg,
The total system power consumption depends on many factors including devices attached to the USB 3.0 Type-A ports.
If you are seeing a potential power issue, I think it is worth considering supplying power through the DC barrel jack (which is rated for 5V/4A, thus 20W).

RyanC1681, 2019-05-12:
Hi,
   I tried both nano and JetBot image with a 5V/2.1A power bank, but the power bank couldn't boot up the system. My power bank has a 2.1 A and 1A port, so can I use both ports at the same time to supply 3.1A power to the Nano? I tried to short J48 with both DC plug and uUSB came from power bank, but it did not work.  Could you recommend a way to supply more current using a power bank? Thanks.

davinderc, 2019-05-17:
@RyanC1681 Is your powerbank really able to provide 3.1 A total? Most powerbanks I've found that fit the provided chassis have two outputs but a maximum combined output of 2.1-2.4 A. I found a powerbank that has a USB C output with Quick Charge 3.0 can provide more, but then I ran into the problem of the QC 3.0 protocol needing to be implemented by whatever device is connected to that.

davinderc, 2019-05-21:
@RyanC1681 I've found that the Nano requires from 0.5-1.5 A of current. During boot up it doesn't seem to go over 1 A, and around 1.5 is peak during object detection, but maybe it peaks a bit higher and my USB power meter wasn't fast enough to spot it. The driver depends on how fast you run the motors, when I set one to 0.9 and the other to 0.75, I was getting around 1.5 A locked motor current, but around 0.5 A free spinning. So ideally we should have 3 A total available, but it's not a constant current requirement.

einsteinagogo, 2019-05-25:
The battery in the BOM is NOW available in the UK via Amazon after discussions with Topstar (INIU)

https://amzn.to/2VRAwWC

davinderc, 2019-05-25:
> The battery in the BOM is NOW available in the UK via Amazon after discussions with Topstar (INIU)
> 
> https://amzn.to/2VRAwWC

Is that an affiliate link I see there?

davinderc, 2019-05-25:
> 
> 
> The battery in the BOM is NOW available in the UK via Amazon after discussions with Topstar (INIU)
> 
> https://amzn.to/2VRAwWC

@einsteinagogo 
Is that an affiliate link I see there?

closed 2022-07-15 by jaybdub