NVIDIA-AI-IOT/jetbot #35: Jupyter fails to initialize camera...most of the times

fciampoli, 2019-04-14:
I have assembled my jetbot with all the recommended hardware, including a Raspberry Pi Camera V2.
The jetbot works and I have been able to follow the examples provided. Unfortunately, most of the times the camera is needed, the notebook fails to activate it. The instruction that creates trouble is "camera = Camera.instance()". When I try to run this cell, jupyter enters Busy mode and never proceeds. This is true most of the times. Sometimes the instance is created successfully and I am able to proceed. This is also true after a fresh reboot.
Any suggestions as to what might be wrong?

bigboss10186, 2019-04-15:
I also have this problem,hope someone can solve this problem.

jaybdub, 2019-04-15:
Hi All,

As a quick check, could you try calling the following from a terminal before executing the code?

```bash
sudo systemctl restart nvargus-daemon
```

We believe occasionally the camera daemon enters a bad state from not shutting down properly.  Please let me know if this resolves your issue.  

We would also like to better understand the context that this occurs in.  Could you all share the steps (from power on) that may be taken to reproduce the issue?  For example,

1. Power on via Micro USB from battery
    > If you use a different battery, or power method, please mention.  Also please list any devices you have connected (ie: USB WiFi or M2 WiFi, HDMI monitor, etc.).
2. Launch ____ jupyter notebook
3. Run cells a, b, c

This will help me reproduce the issue.  Also, when you run into the issue, it's possible the error is contained in the camera daemon or jupyter service log files.  Could you call the following after you encounter the error.

```bash
journalctl -u jetbot_jupyter >> jetbot_jupyter_log.txt
journalctl -u nvargus-daemon >> nvargus_daemon_log.txt
```

This will store the logs in ``jetbot_jupyter_log.txt`` and ``nvargus_daemon_log.txt`` which may be inspected for errors.

Please let me know if you have any questions!  Sorry for the trouble, I will also work on reproducing myself, but any additional information you can provide will be very helpful to us and other JetBot users!

Best,
John

gizmohd, 2019-04-17:
I'm getting these errors especially when attempting to run the object_following/live_demp.ipynb

I've followed your directions and have created the log files in the attached zip file.
[jetbot_logs.zip](https://github.com/NVIDIA-AI-IOT/jetbot/files/3087884/jetbot_logs.zip)

dvillevald, 2019-04-25:
I have the same problem.

phdussud, 2019-04-26:
Same here as well. The logs look identical.

walros007, 2019-04-27:
> same here
found the issue - actually it was a hardware issue, the camera wasn't connected the way it was supposed to be... now initialization works.

jaybdub, 2019-04-27:
Hi walros007, glad to hear you've got that part working!  Please let me know if you encounter further issues.

Still, since multiple people have encountered this, there may be another underlying issue (or we should clarify the Camera connection in hardware setup).  We're looking into it and hope to get back ASAP.

Any additional information is helpful and appreciated!

Thanks!
John

phdussud, 2019-04-27:
I dug a little more into this. While I always get the logs indicating an array access out of bounds from the nvargus_daemon(see gizmohd logs), the camera sometimes works. If it does not, then I restart the netbook kernel and the daemon until it works.

RyanC1681, 2019-04-27:
I didn't see camera working with attach log. Any idea?
[jetbot_jupyter_log.txt](https://github.com/NVIDIA-AI-IOT/jetbot/files/3124112/jetbot_jupyter_log.txt)

[nvargus_daemon_log.txt](https://github.com/NVIDIA-AI-IOT/jetbot/files/3124109/nvargus_daemon_log.txt)

walros007, 2019-04-28:
Hi John,
many thanks for your reply.
i have one more question: How can i install OpenCV 3.4 on my nano?
Many thanks,
Walter

dvillevald, 2019-04-30:
Any luck with this issue?

dvillevald, 2019-05-04:
I replaced the camera and it works ok so it seems in my case something was wrong with the camera (or the connection cable.)

potterhere, 2020-11-29:
solved by running the cmd: 

> sudo systemctl restart nvargus-daemon

jlutzwpi, 2021-05-19:
I tried doing the daemon restart and it's still not working for me....however, I tested the camera directly on the nano and it does work.