NVIDIA-AI-IOT/jetbot #39: Comments on Setup Instructions

wphuhn, 2019-04-18:
After waiting weeks for all the parts to arrive, learning to 3D print, and (re-)learning how to solder, I managed to build my Jetbot and it didn't catch on fire (see https://twitter.com/william_p_huhn/status/1118720548157128705).  I have a few suggestions for changes to the instructions, which I've grouped by section.  Hope these help.

## Hardware Setup ##

* "Tools Needed"
   * Some form of precision jeweler/electronics screwdriver set should be added.  Securing the power and motor wires to the motor driver without them took a considerable amount of effort.
* "Step 2 - Mount motors"
   * It should be mentioned that the motors are attached with the M3 screws *and nuts*.
* "Step 4 - Strip motor driver power cable"
   * It's mentioned to remove the green and white wires, but the final diagram shows them twisted and not removed.
* "Step 10 - Solder header onto PiOLED display"
   * A schematic diagram for which connections should be soldered and which wires go where would be hugely useful for this step and the next.  The camera angle makes it difficult to see.
* "Step 11 - Wire motor driver to PiOLED display"
   * The color contrast between orange and yellow is really bad in the picture.  I had to go look at pictures in subsequent steps to figure out which was which.
   * The instructions reference SDA (orange) and SCL (yellow), which I realize are the proper names, but they're labeled 3 and 5 on the board itself.
* "Step 13 - Attach wide angle sensor"
   * This step should come before "Step 12 - Mount camera", as it's much easier to attach the wide angle sensor without having to account for the angle of the mount on the Jetbot.
   * This is obvious, but the instructions never state to take the camera lens off.
* "Step 15 - Organize wires"
   * The pictures in this section shows the Jetbot's power wires hooked up so that the motor will draw power, which I imagine most people will do even though it's not explicitly stated to do so, but Step 2 of the Software Setup specifically states that we should not power the motor until after the first booting/initial setup.  I think a notice to not power the Jetbot yet would be useful.
* Miscellaneous
   * This is really obvious, but the instructions never state to attach the wheels.  (It looks like it happens sometime between Step 9 and 11.)

## Software Setup ##

* "Step 5 - Install latest software (optional)"
   * On an JetBot SD image that I flashed yesterday night (18 April, 2019), `rsync` was not installed by default, but it was easily installable via `apt`.
   * The user will need to change directory up one level before running `rsync jetbot/notebooks ~/Notebooks`.

## Examples ##

* "Example 2 - Teleoperation"
   * The widget for controlling the Jetbot via my Xbox One controller never showed up, although <https://html5gamepad.com/> was able to detect it.  No error messages or anything.

arraypan, 2019-04-18:
How did you get the wheels, they are still out of stocks

wphuhn, 2019-04-18:
@arraypan I used the 65mm wheels and the associated caster STLs, as they've been in stock for a while (and still appear to be, according to the Adafruit website.)

arraypan, 2019-04-19:
Thanks, I just saw this.

jaybdub, 2019-04-20:
Hi wphuhn,

Thanks for this very thorough feedback!

I think you make a lot of good points.  I'll leave this issue open until we can scope and address these.

Glad to hear you finished building!  Hope you're having fun, if you find any other issues please let us know.

Thanks!
John

coppoljja, 2019-05-07:
Motor controller power cable is the other one than depicted.

closed 2022-07-15 by jaybdub