NVIDIA-AI-IOT/jetbot #40: the 63Gb image is broken

infragilis, 2019-04-19:
it will not boot, after using etcher on a 64Gb card it stops at boot with the following options:

version 00.80.2018.01T210
-continue
-boot recovery kernel
-boot safe mode
and a few more options, keyboard unresponsive

using the standard image and instructions:
https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Create-SD-Card-Image-From-Scratch

works on the same hardware.

wahid, 2019-04-22:
63GB image worked fine for me. Maybe try to load it using etcher again?

infragilis, 2019-04-22:
Multiple attempts, on separate SD cards. Even used older etcher version.

wahid, 2019-04-22:
> Multiple attempts, on separate SD cards. Even used older etcher version.
> […](#)

Maybe just use the standard image and just clone this git and install everything manually? I did that the first time and that worked also.

infragilis, 2019-04-22:
Yup that’s what i did, that worked fine. With some minor issues, but a lot
better :)

Arti11ery, 2019-07-03:
Same issue. Manually installation works for me.

closed 2022-07-15 by jaybdub