NVIDIA-AI-IOT/jetbot #44: PiOLED does not display ipaddress

jomoengineer, 2019-04-24:
Hi,
I am just getting started putting together my JetBot and have loaded the latest JetBot image and installed and configured the WiFi module and installed the PiOLED but I am not seeing anything displayed on the PIOLED after boot.   

NOTE: I have the Nano powered via the barrel jack.

From a forum suggest I had seen, I ran the following which shows some errors.
`journalctl -u jetbot_stats >> jetbot_stats_log.txt
`
Errors seen in the log file
`Apr 24 08:26:20 jetbot sh[9781]: OSError: [Errno 121] Remote I/O error
Apr 24 08:26:22 jetbot systemd[1]: jetbot_stats.service: Main process exited, code=exited, status=1/FAILURE
Apr 24 08:26:22 jetbot systemd[1]: jetbot_stats.service: Failed with result 'exit-code'.
Apr 24 08:26:22 jetbot systemd[1]: jetbot_stats.service: Service hold-off time over, scheduling restart.
Apr 24 08:26:22 jetbot systemd[1]: jetbot_stats.service: Scheduled restart job, restart counter is at 29.
Apr 24 08:26:22 jetbot systemd[1]: Stopped JetBot stats display service.
Apr 24 08:26:22 jetbot systemd[1]: Started JetBot stats display service.
Apr 24 08:26:39 jetbot sh[9877]: Traceback (most recent call last):
Apr 24 08:26:39 jetbot sh[9877]:   File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
Apr 24 08:26:39 jetbot sh[9877]:     "__main__", mod_spec)
Apr 24 08:26:39 jetbot sh[9877]:   File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
Apr 24 08:26:39 jetbot sh[9877]:     exec(code, run_globals)
Apr 24 08:26:39 jetbot sh[9877]:   File "/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/apps/stats.py", line 36, in <module>
Apr 24 08:26:39 jetbot sh[9877]:     disp.begin()
Apr 24 08:26:39 jetbot sh[9877]:   File "/usr/local/lib/python3.6/dist-packages/Adafruit_SSD1306-1.6.2-py3.6.egg/Adafruit_SSD1306/SSD1306.py", line 148, in begin
Apr 24 08:26:39 jetbot sh[9877]:   File "/usr/local/lib/python3.6/dist-packages/Adafruit_SSD1306-1.6.2-py3.6.egg/Adafruit_SSD1306/SSD1306.py", line 292, in _initialize
Apr 24 08:26:39 jetbot sh[9877]:   File "/usr/local/lib/python3.6/dist-packages/Adafruit_SSD1306-1.6.2-py3.6.egg/Adafruit_SSD1306/SSD1306.py", line 129, in command
Apr 24 08:26:39 jetbot sh[9877]:   File "/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.3-py3.6.egg/Adafruit_GPIO/I2C.py", line 116, in write8
Apr 24 08:26:39 jetbot sh[9877]:   File "/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-0.2.3-py3.6.egg/Adafruit_PureIO/smbus.py", line 256, in write_byte_data
Apr 24 08:26:39 jetbot sh[9877]: OSError: [Errno 121] Remote I/O error
Apr 24 08:26:41 jetbot systemd[1]: jetbot_stats.service: Main process exited, code=exited, status=1/FAILURE
Apr 24 08:26:41 jetbot systemd[1]: jetbot_stats.service: Failed with result 'exit-code'.
Apr 24 08:26:41 jetbot systemd[1]: jetbot_stats.service: Service hold-off time over, scheduling restart.
Apr 24 08:26:41 jetbot systemd[1]: jetbot_stats.service: Scheduled restart job, restart counter is at 30.
Apr 24 08:26:41 jetbot systemd[1]: Stopped JetBot stats display service.
Apr 24 08:26:41 jetbot systemd[1]: Started JetBot stats display service.
`

Does this require some sort of initialization?

Thanks,

Jon

jaybdub, 2019-04-25:
Hi jomoengineer,

It seems like the error is in writing over I2C to the display.  This is indicated by the ``[Errno 121] Remote I/O error``.

Is the PiOLED connected correctly to the 40pin header?

Best,
John

jomoengineer, 2019-04-25:
Yes, it is connected, but I'm thinking it is not working.  I've tried it on a RasPi it does not recognize it as well.

I'll have to order another and see if it will work.

Thanks,

Jon

jaybdub, 2019-04-25:
Ah.  Sorry to hear that.  

I'm not sure if you did this before, but as a tip I would recommend testing the PiOLED on the Jetson Nano (without motor driver connected) when you get the new one.

This way you can verify the PiOLED is working on it's own.  Then, double check the motor driver wiring to make sure everything is alright :)

Please let me know if you run into any issues.

Best,
John

jomoengineer, 2019-04-26:
John,

Thanks.  Yeah, I think it was my fault and may have killed it when putting the header on.

The Motor Driver works fine and my bot is Beboppin' around. 

I've ordered a new PiOLED and will try again.  However, I am not too keen on the right angle header since it blocks the Fan Power header, so I intend to use jumper wires instead.

Thanks,

Jon

closed 2022-07-15 by jaybdub