NVIDIA-AI-IOT/jetbot #45: i2c Remote I/O error

diefimov, 2019-04-24:
Hi!

I have finished my jetbot now and trying to run the Basic Motion notebook. I have successfully imported Robot class:

```
from jetbot import Robot
```

but when I run

```
robot = Robot()
```
I got the following error:

```
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-6-b418ad9f6ab3> in <module>
----> 1 robot = Robot()

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in __init__(self, addr, freq, i2c, i2c_bus)
    229         self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
    230         self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
--> 231         self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
    232         self._pwm.setPWMFreq(self._frequency)
    233 

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in __init__(self, address, debug, i2c, i2c_bus)
     57         self.i2c = get_i2c_device(address, i2c, i2c_bus)
     58         logger.debug("Reseting PCA9685 MODE1 (without SLEEP) and MODE2")
---> 59         self.setAllPWM(0, 0)
     60         self.i2c.write8(self.__MODE2, self.__OUTDRV)
     61         self.i2c.write8(self.__MODE1, self.__ALLCALL)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setAllPWM(self, on, off)
     93     def setAllPWM(self, on, off):
     94         "Sets a all PWM channels"
---> 95         self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
     96         self.i2c.write8(self.__ALL_LED_ON_H, on >> 8)
     97         self.i2c.write8(self.__ALL_LED_OFF_L, off & 0xFF)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.3-py3.6.egg/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-0.2.3-py3.6.egg/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    254         # Send the data to the device.
    255         self._select_device(addr)
--> 256         self._device.write(data)
    257 
    258     def write_word_data(self, addr, cmd, val):

OSError: [Errno 121] Remote I/O error
```

Could you please help?
Thanks!

jaybdub, 2019-04-25:
Hi diefimov,

Thanks for reaching out!  It seems that there is an issue communicating with the motor driver.

This error most commonly occurs when the I2C SDA and SCL pins (orange, yellow in diagram) are switched.  Could you verify that these are correct?

Please let me know if that works or if you run into issues still!

Best,
John

diefimov, 2019-04-25:
Hi John,

Thank you for the answer! The problem does relate to the wire connections: I tried to re-solder header to the PiOLED display and it works now.

Thanks,
Dmitry

jaybdub, 2019-04-25:
Awesome!  Glad to hear it's now working :)

Hope you have fun!  Please let me know if you run into any other issues.

John

RyanC1681, 2019-04-29:
Hi,
  I got i2c log as below, but it PiOLED didn't display anything and JetBot example didn't work.
I checked power, I2C wiring are correct and Jupyter connected.  any idea?  
Thanks,
[   64.892046] tegra-i2c 7000c400.i2c: --- register dump for debugging ----     
[   64.901488] tegra-i2c 7000c400.i2c: I2C_CNFG - 0x22c00                       
[   64.906949] tegra-i2c 7000c400.i2c: I2C_PACKET_TRANSFER_STATUS - 0x10001     
[   64.914453] tegra-i2c 7000c400.i2c: I2C_FIFO_CONTROL - 0xe0                  
[   64.920078] tegra-i2c 7000c400.i2c: I2C_FIFO_STATUS - 0x800070               
[   64.925934] tegra-i2c 7000c400.i2c: I2C_INT_MASK - 0x7c                      
[   64.931720] tegra-i2c 7000c400.i2c: I2C_INT_STATUS - 0x0                     
[   64.937462] tegra-i2c 7000c400.i2c: i2c transfer timed out addr: 0x3c        
[   94.066315] tegra-i2c 7000c400.i2c: pio timed out addr: 0x3c tlen:16 rlen:0  
[   94.073490] tegra-i2c 7000c400.i2c: --- register dump for debugging ----     
[   94.081839] tegra-i2c 7000c400.i2c: I2C_CNFG - 0x22c00                       
[   94.087924] tegra-i2c 7000c400.i2c: I2C_PACKET_TRANSFER_STATUS - 0x10001     
[   94.111738] tegra-i2c 7000c400.i2c: I2C_FIFO_CONTROL - 0xe0                  
[   94.122529] tegra-i2c 7000c400.i2c: I2C_FIFO_STATUS - 0x800070               
[   94.129228] tegra-i2c 7000c400.i2c: I2C_INT_MASK - 0x7c                      
[   94.137909] tegra-i2c 7000c400.i2c: I2C_INT_STATUS - 0x0                     
[   94.144178] tegra-i2c 7000c400.i2c: i2c transfer timed out addr: 0x3c

RyanC1681, 2019-05-01:
I also try "sudo i2cdetect -y 1" without motor drive board, but it didn't detect any I2C device.
Any idea?
Thanks,

jaybdub, 2019-05-02:
Hi RyanC,

Does the motor driver work when connected to the Nano directly?  I'm wondering if there is just an issue with the PiOLED hardware.

John

RyanC1681, 2019-05-02:
Hi, John:
 No, the motor drive light up, but it didn't drive motor when I tried to run basic_motion and PiOLED didn't display anything.  Here is the i2c bus as below, which one connect to OLED and motor drive?
sudo i2cdetect -l
i2c-3	i2c       	7000c700.i2c                    	I2C adapter
i2c-1	i2c       	7000c400.i2c                    	I2C adapter
i2c-6	i2c       	Tegra I2C adapter               	I2C adapter
i2c-4	i2c       	7000d000.i2c                    	I2C adapter
i2c-2	i2c       	7000c500.i2c                    	I2C adapter
i2c-0	i2c       	7000c000.i2c                    	I2C adapter
i2c-5	i2c       	7000d100.i2c                    	I2C adapter

RyanC1681, 2019-05-02:
Hi, quick update, I can make motor drive work to drive motor now, but it still couldn't display PiOLED or found 0x3c i2c device as below:
sudo i2cdetect -y 1 #scope shown I2C SCL, SDA are toggle
Warning: Can't use SMBus Quick Write command, will skip some addresses

sudo python3 stats.py  #scope didn't show I2C SCL, SDA toggle
Traceback (most recent call last):
  File "stats.py", line 30, in <module>
    from board import SCL, SDA
ModuleNotFoundError: No module named 'board'

RyanC1681, 2019-05-03:
Hi, Does Jetson Nano/JetBot support PiOLED I2C 0.91inch OLED 128x32 SSD1306 White for RPI Raspberry Pi?  Is there other way to test PiOLED before I order another one because ADA3527 is out of stock?
Thanks,

terechsan, 2019-08-07:
I get this error randomly after initializing robot, and controlling it with collision_avoidance module.

abhinavos7a, 2019-09-10:
I am trying to set up SparkFun JetBot AI Kit  with NVIDIA Jetson Nano.
I am not able to run the 1st experiment called Basic motion through python notebook.

I have made sure the soldering is correct on the motor driver board, I can see red LED blinking every 2.5 seconds (working normally).
The connection from Micro OLED to motor driver is correct (Yellow to "SCL," Blue to "SDA," Black to "GND")

I also tested motors separately (They work by powering through my AA batteries)

I don't get any errors while initializing the robot and calling methods right and left by passing speed arguments. It's just that the robot doesn't do anything.

Can somebody help how can I debug my setup ?

zoolyka, 2019-10-02:
I've just played with an adafruit motor driver (same as the one used in the jetbot project) using latest jetpack.

A few things to mention that can make it easier to set it up:

- `sudo i2cdetect -l` is almost useless, the output wont change once you connected the motor driver
- `sudo i2cdetect -y 1` same thing, use `sudo i2cdetect -y -r 1` instead, if the motor driver is recognised it will appear on address 60 and 70 (without the -r option it will not test these addresses)

If the complete jetbot bundle is not required, just want to rotate the motors, then https://github.com/adafruit/Adafruit-Motor-HAT-Python-Library/ works well, but have to edit a single line before build & install.

In Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py line 227 change:
`def __init__(self, addr = 0x60, freq = 1600, i2c=None, i2c_bus=None):`
to:
`def __init__(self, addr = 0x60, freq = 1600, i2c=None, i2c_bus = 1):`

Otherwise it wont recognise the Nano's default bus, which is 1.

Gavincko227, 2020-01-16:
I am trying to run the first example called basic motion (https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Examples) on a kit I purchased from seedstudio (https://www.seeedstudio.com/Seeedstudio-JetBot-Smart-Car-Powered-by-NVIDIA-Jetson-Nano-p-4054.html). I am using this kit in combination with the Jetson Nano computer. The oLED is working and the red light from the motor driver is also on. However I am having trouble getting the motors to work. I have soldered on the connections to the motor, and my other cables seem secure. As i follow through the tutorial, I am get the following error: 

`OSError                                   Traceback (most recent call last)
<ipython-input-3-e352ea409cdb> in <module>
----> 1 robot.left(speed=0.3)

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/robot.py in left(self, speed)
     38     def left(self, speed=1.0):
     39         self.left_motor.value = -speed
---> 40         self.right_motor.value = speed
     41 
     42     def right(self, speed=1.0):

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in __set__(self, obj, value)
    582             raise TraitError('The "%s" trait is read-only.' % self.name)
    583         else:
--> 584             self.set(obj, value)
    585 
    586     def _validate(self, obj, value):

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in set(self, obj, value)
    571             # we explicitly compare silent to True just in case the equality
    572             # comparison above returns something other than True/False
--> 573             obj._notify_trait(self.name, old_value, new_value)
    574 
    575     def __set__(self, obj, value):

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in _notify_trait(self, name, old_value, new_value)
   1203             new=new_value,
   1204             owner=self,
-> 1205             type='change',
   1206         ))
   1207 

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in notify_change(self, change)
   1208     def notify_change(self, change):
   1209         """Notify observers of a change event"""
-> 1210         return self._notify_observers(change)
   1211 
   1212     def _notify_observers(self, event):

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in _notify_observers(self, event)
   1245                 c = getattr(self, c.name)
   1246 
-> 1247             c(event)
   1248 
   1249     def _add_notifiers(self, handler, name, type):

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/motor.py in _observe_value(self, change)
     22     @traitlets.observe('value')
     23     def _observe_value(self, change):
---> 24         self._write_value(change['new'])
     25 
     26     def _write_value(self, value):

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/motor.py in _write_value(self, value)
     32             self._motor.run(Adafruit_MotorHAT.FORWARD)
     33         else:
---> 34             self._motor.run(Adafruit_MotorHAT.BACKWARD)
     35 
     36     def _release(self):

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in run(self, command)
    202         if (command == Adafruit_MotorHAT.BACKWARD):
    203             self.MC.setPin(self.IN1pin, 0)
--> 204             self.MC.setPin(self.IN2pin, 1)
    205         if (command == Adafruit_MotorHAT.RELEASE):
    206             self.MC.setPin(self.IN1pin, 0)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in setPin(self, pin, value)
    240             self._pwm.setPWM(pin, 0, 4096)
    241         if (value == 1):
--> 242             self._pwm.setPWM(pin, 4096, 0)
    243 
    244     def getStepper(self, steps, num):

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setPWM(self, channel, on, off)
     87         "Sets a single PWM channel"
     88         self.i2c.write8(self.__LED0_ON_L+4*channel, on & 0xFF)
---> 89         self.i2c.write8(self.__LED0_ON_H+4*channel, on >> 8)
     90         self.i2c.write8(self.__LED0_OFF_L+4*channel, off & 0xFF)
     91         self.i2c.write8(self.__LED0_OFF_H+4*channel, off >> 8)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.3-py3.6.egg/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-0.2.3-py3.6.egg/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    254         # Send the data to the device.
    255         self._select_device(addr)
--> 256         self._device.write(data)
    257 
    258     def write_word_data(self, addr, cmd, val):

OSError: [Errno 121] Remote I/O error`

I was wondering if someone would be able to help me resolve this issue.

joshdunk98, 2020-01-29:
@jaybdub Is the pioled necessary for the jetbot? I am planning to build on the code to develop my own object detection and facial recognition code, so I am in no need of using the pioled display. I already have access to the jupyter notebooks and everything, so is there a way to disable the use of the ic2 in the code or remove the REMOTE I/O ERROR?

chelpuri, 2020-02-27:
def __init__(self, addr = 0x60, freq = 1600, i2c=None, i2c_bus=None):
        self._frequency = freq
        self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
        self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
        **self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)**
        self._pwm.setPWMFreq(self._frequency)

i am using servo shield pca9685 16 channel pwm driver i2c ,after running python code in rasp pi
i am getting  OSError: [Errno 121] Remote I/O error can u help me in solving how to change address of i2c shield

bethalageetachandraraju, 2020-03-31:
I tried all but the same error in jetbot while I was running " robot = Robot() "
I got the following error:

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-6-b418ad9f6ab3> in <module>
----> 1 robot = Robot()

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in __init__(self, addr, freq, i2c, i2c_bus)
    229         self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
    230         self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
--> 231         self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
    232         self._pwm.setPWMFreq(self._frequency)
    233 

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in __init__(self, address, debug, i2c, i2c_bus)
     57         self.i2c = get_i2c_device(address, i2c, i2c_bus)
     58         logger.debug("Reseting PCA9685 MODE1 (without SLEEP) and MODE2")
---> 59         self.setAllPWM(0, 0)
     60         self.i2c.write8(self.__MODE2, self.__OUTDRV)
     61         self.i2c.write8(self.__MODE1, self.__ALLCALL)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setAllPWM(self, on, off)
     93     def setAllPWM(self, on, off):
     94         "Sets a all PWM channels"
---> 95         self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
     96         self.i2c.write8(self.__ALL_LED_ON_H, on >> 8)
     97         self.i2c.write8(self.__ALL_LED_OFF_L, off & 0xFF)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.3-py3.6.egg/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-0.2.3-py3.6.egg/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    254         # Send the data to the device.
    255         self._select_device(addr)
--> 256         self._device.write(data)
    257 
    258     def write_word_data(self, addr, cmd, val):

OSError: [Errno 121] Remote I/O error.

I checked the pin connections several times but same error.
Please anyone help me.

229477172, 2020-04-11:
I ran into the same problem when I executed “robot = Robot()”.
But I have not installed the motor and there is no wiring, is it right to connect the wire installed motor will not have this problem?

OSError                                   Traceback (most recent call last)
<ipython-input-5-b418ad9f6ab3> in <module>
----> 1 robot = Robot()

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in __init__(self, addr, freq, i2c, i2c_bus)
    229         self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
    230         self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
--> 231         self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
    232         self._pwm.setPWMFreq(self._frequency)
    233 

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in __init__(self, address, debug, i2c, i2c_bus)
     57         self.i2c = get_i2c_device(address, i2c, i2c_bus)
     58         logger.debug("Reseting PCA9685 MODE1 (without SLEEP) and MODE2")
---> 59         self.setAllPWM(0, 0)
     60         self.i2c.write8(self.__MODE2, self.__OUTDRV)
     61         self.i2c.write8(self.__MODE1, self.__ALLCALL)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setAllPWM(self, on, off)
     93     def setAllPWM(self, on, off):
     94         "Sets a all PWM channels"
---> 95         self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
     96         self.i2c.write8(self.__ALL_LED_ON_H, on >> 8)
     97         self.i2c.write8(self.__ALL_LED_OFF_L, off & 0xFF)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.4-py3.6.egg/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-1.0.4-py3.6.egg/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    266         # Send the data to the device.
    267         self._select_device(addr)
--> 268         self._device.write(data)
    269 
    270     def write_word_data(self, addr, cmd, val):

OSError: [Errno 121] Remote I/O error

jyotiitbhu, 2020-07-06:
i am also getting same error. I checked connections like millions of times, wires are connected correctly. 
How to debug this?

beni1000, 2020-10-19:
I am getting the same Error tried all the steps above.. are there any new solutions for this problem yet?
thx in advance

tdwilhelm, 2021-03-15:
I started working on this project last week implementing the full ROS Jetbot solution ( https://github.com/dusty-nv/jetbot_ros) and I'm having the same issue. All instructions followed on https://github.com/dusty-nv/jetbot_ros with no errors.

I tried many things but have not been able to resolve. The sudo i2cdetect -y -r 1 command showed the OLED display at 0X30 but not the Adafruit driver board.

I tried just connecting the board to the nano pins directly, without the OLED, no luck. 

I changed the Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py line 227 change:
def __init__(self, addr = 0x60, freq = 1600, i2c=None, i2c_bus=None):
to:
def __init__(self, addr = 0x60, freq = 1600, i2c=None, i2c_bus = 1): no luck.

I check the wiring based on , no luck....

Checked all solder points.

The power light is on on the Adafruit Board

JetPack 4.5.1 
Ubuntu 18.04 LTS
ROS Melodic 

Using GPIO Nano Pins 1,3,5,6

Adafruit Board - DC MOTOR + STEPPER FEATHERWING ADD-ON FOR ALL FEATHER BOARDS
https://www.adafruit.com/product/2927 

ros@jetson:~/workspace/catkin_ws/src/jetbot_ros/scripts$ sudo i2cdetect -y -r 1
[sudo] password for ros: 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    

RUN node:
rosrun jetbot_ros jetbot_motors.py

CODE for jetbot_motors.py:

#!/usr/bin/env python
import rospy
import time

from Adafruit_MotorHAT import Adafruit_MotorHAT
from std_msgs.msg import String

# sets motor speed between [-1.0, 1.0]
def set_speed(motor_ID, value):
max_pwm = 115.0
speed = int(min(max(abs(value * max_pwm), 0), max_pwm))

if motor_ID == 1:
motor = motor_left
elif motor_ID == 2:
motor = motor_right
else:
rospy.logerror('set_speed(%d, %f) -> invalid motor_ID=%d', motor_ID, value, motor_ID)
return

motor.setSpeed(speed)

if value > 0:
motor.run(Adafruit_MotorHAT.FORWARD)
else:
motor.run(Adafruit_MotorHAT.BACKWARD)

# stops all motors
def all_stop():
motor_left.setSpeed(0)
motor_right.setSpeed(0)

motor_left.run(Adafruit_MotorHAT.RELEASE)
motor_right.run(Adafruit_MotorHAT.RELEASE)

# directional commands (degree, speed)
def on_cmd_dir(msg):
rospy.loginfo(rospy.get_caller_id() + ' cmd_dir=%s', msg.data)

# raw L/R motor commands (speed, speed)
def on_cmd_raw(msg):
rospy.loginfo(rospy.get_caller_id() + ' cmd_raw=%s', msg.data)

# simple string commands (left/right/forward/backward/stop)
def on_cmd_str(msg):
rospy.loginfo(rospy.get_caller_id() + ' cmd_str=%s', msg.data)

if msg.data.lower() == "left":
set_speed(motor_left_ID,  -1.0)
set_speed(motor_right_ID,  1.0) 
elif msg.data.lower() == "right":
set_speed(motor_left_ID,   1.0)
set_speed(motor_right_ID, -1.0) 
elif msg.data.lower() == "forward":
set_speed(motor_left_ID,   1.0)
set_speed(motor_right_ID,  1.0)
elif msg.data.lower() == "backward":
set_speed(motor_left_ID,  -1.0)
set_speed(motor_right_ID, -1.0)  
elif msg.data.lower() == "stop":
all_stop()
else:
rospy.logerror(rospy.get_caller_id() + ' invalid cmd_str=%s', msg.data)

# initialization
if __name__ == '__main__':

# setup motor controller
motor_driver = Adafruit_MotorHAT(i2c_bus=1)

motor_left_ID = 1
motor_right_ID = 2

motor_left = motor_driver.getMotor(motor_left_ID)
motor_right = motor_driver.getMotor(motor_right_ID)

# stop the motors as precaution
all_stop()

# setup ros node
rospy.init_node('jetbot_motors')

rospy.Subscriber('~cmd_dir', String, on_cmd_dir)
rospy.Subscriber('~cmd_raw', String, on_cmd_raw)
rospy.Subscriber('~cmd_str', String, on_cmd_str)

# start running
rospy.spin()

# stop motors before exiting
all_stop()
---

ros@jetson:~/workspace/catkin_ws/src/jetbot_ros/scripts$ rosrun jetbot_ros jetbot_motors.py
Traceback (most recent call last):
  File "/home/ros/workspace/catkin_ws/src/jetbot_ros/scripts/jetbot_motors.py", line 75, in <module>
    motor_driver = Adafruit_MotorHAT(i2c_bus=1)
  File "/home/ros/.local/lib/python2.7/site-packages/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py", line 231, in __init__
    self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
  File "/home/ros/.local/lib/python2.7/site-packages/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py", line 59, in __init__
    self.setAllPWM(0, 0)
  File "/home/ros/.local/lib/python2.7/site-packages/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py", line 95, in setAllPWM
    self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
  File "/home/ros/.local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/home/ros/.local/lib/python2.7/site-packages/Adafruit_PureIO/smbus.py", line 268, in write_byte_data
    self._device.write(data)
IOError: [Errno 121] Remote I/O error

             
Any thoughts? Beyond frustrating!

tdwilhelm, 2021-03-26:
Motor board was BAD. Adafruit sent me a new one. That worked.

Pranav-India, 2021-05-15:
@tdwilhelm Hi can you tell me how did you figure out it was motor board issue?

tdwilhelm, 2021-05-15:
Hi Pranav_India, I tried just about everything, the only thing left was to get a new board and test. I got the new board and the I2c device then showed showing up when using the sudo i2cdetect -y -r 1 command.

I went on to discover that there were still other issues!  the test code given didn't work. I gave up on the board and this version of jetbot as described https://jetbot.org/master/. I went to the Waveshare JetBot AI Kit instead.  It worked great!

Pranav-India, 2021-05-17:
Hi @tdwilhelm Thank you for your response. I will try to find out what is causing the error on my system. When I run the sudo  i2cdetect -y -r 1 command I am able to see the i2c device there but still get the **OSError: [Errno 121] Remote I/O error** error when I try to instantiate the robot object

santaimpersonator, 2021-06-22:
@abhinavos7a For the SparkFun Kit, you need to use their pre-built image as mentioned in the hookup guide. You can follow the [instructions here](https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/4-software-setup-guide-from-nvidia).

Taitan2010, 2021-07-25:
**Please solve my problem as well. I am a Japanese living near the cycle race course in Tokyo Olympics. I am facing the similar problems as my seniors.  I always get the following error for "robot =Robot ()"  It's connected to both the motor and the display, right? In the program, there are no places to specify the I2C address relevant to each device ...**

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-b418ad9f6ab3> in <module>
----> 1 robot = Robot()

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in __init__(self, addr, freq, i2c, i2c_bus)
    229         self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
    230         self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
--> 231         self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
    232         self._pwm.setPWMFreq(self._frequency)
    233 

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in __init__(self, address, debug, i2c, i2c_bus)
     57         self.i2c = get_i2c_device(address, i2c, i2c_bus)
     58         logger.debug("Reseting PCA9685 MODE1 (without SLEEP) and MODE2")
---> 59         self.setAllPWM(0, 0)
     60         self.i2c.write8(self.__MODE2, self.__OUTDRV)
     61         self.i2c.write8(self.__MODE1, self.__ALLCALL)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setAllPWM(self, on, off)
     93     def setAllPWM(self, on, off):
     94         "Sets a all PWM channels"
---> 95         self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
     96         self.i2c.write8(self.__ALL_LED_ON_H, on >> 8)
     97         self.i2c.write8(self.__ALL_LED_OFF_L, off & 0xFF)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    266         # Send the data to the device.
    267         self._select_device(addr)
--> 268         self._device.write(data)
    269 
    270     def write_word_data(self, addr, cmd, val):

OSError: [Errno 121] Remote I/O error

**Since thinking "Isn't it connected to the device?", I checked as follows, but it seems that there is no problem.**

PS H:\> ssh jetbot@192.168.55.1
jetbot@192.168.55.1's password:
Welcome to Ubuntu 18.04.3 LTS (GNU/Linux 4.9.140-tegra aarch64)Welcome to Ubuntu 18.04.3 LTS (GNU/Linux 4.9.140-tegra aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

This system has been minimized by removing packages and content that are
not required on a system that users do not log into.

To restore this content, you can run the 'unminimize' command.

554 packages can be updated.
446 updates are security updates.

Last login: Sun Jul 11 16:34:25 2021 from 192.168.2.182
[... 70 lines cut ...]
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX 00 00 00 00 1e 00    XXXXXXXXXX....?.

jetbot@jetson-4-3:~$ sudo i2cdump -y 1 0x3c
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
10: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
20: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
30: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
40: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
50: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
60: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
70: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
80: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
90: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
a0: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
b0: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
c0: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
d0: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
e0: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????
f0: 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03    ????????????????

**Thank you for your support in advance!!!**

RyanC1681, 2021-07-25:
Did you see I2C display working? Have PCA9685 or motor ever worked?
If you can see I2C devices(0x40, 0x70) then I2C works fine, then the problem could be PCA9685 or motor connection issue.
As last comment above if you use SparkFun Kit, you need to use their pre-built image as mentioned in the hookup guide. You can follow the instructions here.https://github.com/NVIDIA-AI-IOT/jetbot/issues/45#issuecomment-866084261

Taitan2010, 2021-07-26:
Dear Ryan
Thank you for your comment. 
I2C display is working without problems, it shows the required information. On the other hand, the motor driver which I have been using is the following one, WAVESHARE Motor Driver HAT.https://www.waveshare.com/wiki/Motor_Driver_HAT
I succeeded in moving the motor on a trial basis with this driver, however, have not tried such as changing the rotation speed or stopping, except starting the motor. Therefore, I'm not sure if this test fully evaluated the availability of motor driver. 
Do you think there is a problem with this motor driver?

     ----- Original Message -----

   

As last comment above if you use SparkFun Kit, you need to use their pre-built image as mentioned in the hookup guide. You can follow the instructions here.#45 (comment)—