NVIDIA-AI-IOT/jetbot #46: rsync: command not found

jomoengineer, 2019-04-25:
Hi,

I'm going through the Software Setup for the JetBot and at under Section 5 step 5 it mentions to run rsync to replace the old notebooks but I am getting an error that rsync is not found

This is with image: jetbot_image_v0p3p0.zip

Ex:
<pre><code>
$ rsync jetbot/notebooks ~/Notebooks

Thanks,

Jon
-bash: rsync: command not found
</code></pre>

jaybdub, 2019-04-25:
Hi Jomoengineer,

Thanks for reaching out!  Sorry, we forgot to include one step in the software setup, which is to install the ``rsync`` command.

You should be able to install it by calling

```bash
sudo apt-get install rsync
```

From a terminal.  Please let me know if you run into any issues, and thanks for pointing this out!

Best,
John

idcrook, 2019-04-26:
I created a wiki "patch" for this issue, which I encountered independently: #48

jomoengineer, 2019-04-26:
> Hi Jomoengineer,
> 
> Thanks for reaching out! Sorry, we forgot to include one step in the software setup, which is to install the `rsync` command.
> 
> You should be able to install it by calling
> 
> ```shell
> sudo apt-get install rsync
> ```
> 
> From a terminal. Please let me know if you run into any issues, and thanks for pointing this out!
> 
> Best,
> John

John,

No worries. 

Yes, that is what I ended up doing.

Thanks,

Jon

jomoengineer, 2019-04-26:
> I created a wiki "patch" for this issue, which I encountered independently: #48

Okay, thanks.

idcrook, 2019-04-26:
It shouldn't hurt to do an `update` first. the following one-liner does it simply:

`sudo apt update && sudo apt install -y rsync`

jomoengineer, 2019-04-27:
One other note regarding rsync.  I notice when running this as listed does not update the existing ~/Notebooks files.   To get it to update the files I had to run the following:

<pre><code>
$ rsync -rv   jetbot/notebooks/* /home/jetbot/Notebooks/
</code></pre>

closed 2022-07-15 by jaybdub