NVIDIA-AI-IOT/jetbot #47: Camera (streaming) does not work

dvillevald, 2019-04-25:
I am trying to run a data_collection notebook and have an issue with the camera - the cell which creates a camera instance takes forever to run and, when/if it is executed, the streaming only works for few seconds (if work at all) and then the image becomes still and does not change. Do you have any suggestions on how to resolve it? Thank you in advance.

jaybdub, 2019-04-25:
Hi dvillevald,

Thanks for reaching out!  We have encountered camera freezing with the object detection notebook which were are looking into, but have not been able to reproduce for the data_collection notebook.

Could you try restarting the notebook and executing the following from a terminal before initializing the Camera?

```bash
sudo systemctl restart nvargus-daemon
```

Also, could you provide the following information (which will be helpful in tracking this problem down).

1.  Are you using the listed battery pack?  If so, roughly what is the battery %.
2.  Do you have any other open jupyter notebooks?  The camera can only be initialized in a single one at a time.
3.  Is the Jetson Nano in MAXN or 5W mode?  You can determine this by calling the following from a terminal

    ```bash
    nvpmodel -q
    ```

Thank you!
John

dvillevald, 2019-04-26:
Hi John,

Thanks a lot for looking into this! 

Yes, I tried `sudo systemctl restart nvargus-daemon`  before initializing the camera but it does not seem to help. 

_1. Are you using the listed battery pack? If so, roughly what is the battery %._ 
Yes, I use the pack mentioned in the bills of materials (100% charged). I also observed the same behavior with (stationary) power supply.  

_2. Do you have any other open jupyter notebooks? The camera can only be initialized in a single one at a time._ 
I only run one notebook data_collection - I see only one kernel running.

_3. Is the Jetson Nano in MAXN or 5W mode?_ 
I got the following when I run `nvpmodel -q`  and it shows 5W.
[image]

In case it might help - 

- I have the issue described above when I use Chrome.  When I try to connect from Internet Explorer and run the same data_collection notebook, I cannot initialize the camera - after trying to run the cell initializing the camera I get **Error creating widget: could not find the model**
- I am on Windows 10 (with NVIDIA GeForce GTX 1050 Ti.)

Thank you again for helping!
Dmitri

jaybdub, 2019-04-26:
Hi Dmitri,

Thanks for sharing, this is helpful.  The error in screenshot you shared indicates the failure occurs when initializing the camera on the robot.  If that's the case, the error may be contained in the camera daemon log or the jupyter kernel log.  Can you call the following

```bash
journalctl -u jetbot_jupyter >> jupyter_log.txt
journalctl -u nvargus-daemon >> nvargus_log.txt
```

This will dump the logs into two files ``jupyter_log.txt`` and ``nvargus_log.txt``.  You can then download these files to your computer.  Could you share these logs with me?

Thank you!
John

dvillevald, 2019-04-27:
Hi John,

Thank you for following up! I am attaching the logs. Please let me know if you want me to run more tests. 
[nvargus_log.txt](https://github.com/NVIDIA-AI-IOT/jetbot/files/3123322/nvargus_log.txt)
[jupyter_log.txt](https://github.com/NVIDIA-AI-IOT/jetbot/files/3123323/jupyter_log.txt)

Thank you.
Dmitri

dvillevald, 2019-05-03:
Any suggestions?

dvillevald, 2019-05-04:
I replaced the camera and it works ok so it seems something was wrong with the camera (or the connection cable.)

Thank you for looking into this!

closed 2019-05-04 by dvillevald