NVIDIA-AI-IOT/jetbot #48: Wiki: Software-Setup.md - update "Replace the old notebook" instruction

idcrook, 2019-04-26:
I made a fix to Software-Setup wiki page for "`rsync: not found`" [here](https://github.com/idcrook/jetbot-wiki/commit/c7208989948f122417e8b9586c7809692f92cac6#diff-4b85e61b8f58c9c1addb5af9f05e0c00R70) in a clone of the wiki repo.
As I cannot find a better way to update wiki I created this issue as an invitation to merge my changes.

Here's the salient snippet:

```diff
 3. Replace the old notebooks with the new notebooks by entering
     ```bash
-    rsync jetbot/notebooks ~/Notebooks
+    rsync -avr notebooks ~/Notebooks
     ```
+   If you get the following error, install `rsync` and try again.
+     ```bash
+     jetbot@jetbot:~/jetbot$ rsync jetbot/notebooks ~/Notebooks
+     bash: rsync: command not found
+     jetbot@jetbot:~/jetbot$ sudo apt install rsync
+     ```
```

jaybdub, 2019-04-26:
Hi idcrook,

I've added ``sudo apt-get install rsync`` to the wiki.

Thanks!
John

dvillevald, 2019-04-26:
When I run ` jetbot@jetbot:~/jetbot$ rsync -avr notebooks ~/Notebooks` 
it does not replace the old (individual) folders with examples. Instead it creates a new subfolder named **notebooks** in folder **Notebooks**.

Should not it include the trailing slash after **notebooks**?:
`jetbot@jetbot:~/jetbot$ rsync -avr notebooks/ ~/Notebooks`

idcrook, 2019-04-26:
Thanks @jaybdub 
As it is,it still doesn't work. `rsync` cmd as written only does top-level dir, not recursively (subdirs).  
 
```shell
$ rsync notebooks ~/Notebooks
skipping directory notebooks
```

Also, it does not follow existing flow there, as it is written for wrong working directory. It should instead be, as @dvillevald corrected above:

```shell
rsync -avr notebooks/. ~/Notebooks
```

I've updated my [wiki clone](https://github.com/idcrook/jetbot-wiki)'s [Software-Setup.md](https://github.com/idcrook/jetbot-wiki/blob/master/Software-Setup.md#step-5---install-latest-software-optional) to the following:

```
3. Replace the old notebooks with the new notebooks by entering
    ```bash
    rsync -avr notebooks/. ~/Notebooks
    ```
   If you get the following error, install `rsync` and try again.
     ```bash
     jetbot@jetbot:~/jetbot$ rsync -avr notebooks/. ~/Notebooks
     bash: rsync: command not found
     jetbot@jetbot:~/jetbot$ sudo apt install -y rsync
     ```
```

idcrook, 2019-05-05:
as mentioned in https://github.com/NVIDIA-AI-IOT/jetbot/issues/46#issuecomment-487268998, the `rsync` command won't work as is. It will now just see that directory is there and "skip".

It should be something that includes `-a` or `-r` switch to recursively `rsync` subdirectories:

```
rsync -av notebooks/. ~/Notebooks
```

idcrook, 2019-05-16:
The `rsync` command is still broken on the Wiki.

However, I am going to close this issue:
1. On GitHub the ease of updating a "wiki" is completely not in the [spirit of wiki](http://wiki.c2.com/?OneMinuteWiki); GitHub should either fix this aspect or call it something else
2. A search of issues for rsync will turn up this issue.

closed 2019-05-16 by idcrook