NVIDIA-AI-IOT/jetbot #49: NVPM ERROR when running nvpmodel -q

jomoengineer, 2019-04-26:
When running 'nvpmodel -q' in Step 6 - Configure power mode from the wiki NVPM ERROR is seen.

Ex:
<pre><code>
jetbot@jetbot:~$ sudo nvpmodel -m1
jetbot@jetbot:~$ nvpmodel -q
NV Power Mode: 5W
1
NVPM ERROR: Error opening /sys/kernel/nvpmodel_emc_cap/emc_iso_cap: 13
NVPM ERROR: failed to read PARAM EMC: ARG MAX_FREQ: PATH /sys/kernel/nvpmodel_emc_cap/emc_iso_cap
</code></pre>

I believe it needs to be run as sudo.

This does not show an error.
Ex: 
<pre><code>
jetbot@jetbot:~$ sudo nvpmodel -q
NV Power Mode: 5W
1
</pre></code>

jaybdub, 2019-04-28:
Hi jomoengineer,

Thanks for pointing this out.  I'll test this change and update the Wiki.

Best,
John

colabora001, 2020-05-04:
The wiki haven't been updated. FYI

Kaju-Bubanja, 2020-10-19:
More than a year later and the issue still exists. I came here from: https://github.com/dusty-nv/jetson-inference/issues/334 and the nano crashes regularly

closed 2022-07-15 by jaybdub