NVIDIA-AI-IOT/jetbot #50: iwlwifi - Microcode SW error detected after upgrade

jomoengineer, 2019-04-26:
After running 'sudo apt-get update && sudo apt-get upgrade, the Wif-Fi connection stopped working and I was not able to connect remotely to the Nano - JetBot.  Booting the board with an HDMI connection and barrel jack power adapter, I see the following iwlwifi errors from dmesg.
NOTE: The WiFi module is a : Intel Dual Band Wireless-Ac 8265 W/Bt

<pre><code>
[   12.700367] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-8265-23.ucode failed with error -2
[   12.711193] iwlwifi 0000:01:00.0: Falling back to user helper
[   12.836779] iwlwifi 0000:01:00.0: loaded firmware version 22.391740.0 op_mode iwlmvm
[   12.964357] Bluetooth: hci0: Bootloader revision 0.0 build 26 week 38 2015
[   12.965623] Bluetooth: hci0: Device revision is 16
[   12.965627] Bluetooth: hci0: Secure boot is enabled
[   12.965631] Bluetooth: hci0: OTP lock is enabled
[   12.965633] Bluetooth: hci0: API lock is enabled
[   12.965636] Bluetooth: hci0: Debug lock is disabled
[   12.965640] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[   12.967338] usbcore: registered new interface driver btusb
[   12.978468] iwlwifi 0000:01:00.0: Detected Intel(R) Dual Band Wireless AC 8265, REV=0x230
[   12.978990] Bluetooth: hci0: Found device firmware: intel/ibt-12-16.sfi
[   12.983818] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.985051] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   13.117997] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'
[   13.119929] thermal thermal_zone6: failed to read out thermal zone (-5)
[   13.133005] thermal thermal_zone6: Registering thermal zone thermal_zone6 for type iwlwifi
[   13.152191] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.152205] Bluetooth: BNEP socket layer initialized
[   13.692157] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   13.696068] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   13.696696] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   13.759033] iwlwifi 0000:01:00.0: Microcode SW error detected.  Restarting 0x2000000.
[   13.766915] iwlwifi 0000:01:00.0: CSR values:
[   13.771332] iwlwifi 0000:01:00.0: (2nd byte of CSR_INT_COALESCING is CSR_INT_PERIODIC_REG)
[   13.779739] iwlwifi 0000:01:00.0:        CSR_HW_IF_CONFIG_REG: 0X00489000
[   13.786663] iwlwifi 0000:01:00.0:          CSR_INT_COALESCING: 0X0000ff40
[   13.793604] iwlwifi 0000:01:00.0:                     CSR_INT: 0X10000000
[   13.800519] iwlwifi 0000:01:00.0:                CSR_INT_MASK: 0X00000000
[   13.807439] iwlwifi 0000:01:00.0:           CSR_FH_INT_STATUS: 0X00000000
[   13.814351] iwlwifi 0000:01:00.0:                 CSR_GPIO_IN: 0X00000010
[   13.821270] iwlwifi 0000:01:00.0:                   CSR_RESET: 0X00000000
[   13.828199] iwlwifi 0000:01:00.0:                CSR_GP_CNTRL: 0X08040005
[   13.835128] iwlwifi 0000:01:00.0:                  CSR_HW_REV: 0X00000230
[   13.842055] iwlwifi 0000:01:00.0:              CSR_EEPROM_REG: 0Xd55555d5
[   13.848979] iwlwifi 0000:01:00.0:               CSR_EEPROM_GP: 0Xd55555d5
[   13.855890] iwlwifi 0000:01:00.0:              CSR_OTP_GP_REG: 0Xd55555d5
[   13.862803] iwlwifi 0000:01:00.0:                 CSR_GIO_REG: 0X001f0042
[   13.869713] iwlwifi 0000:01:00.0:            CSR_GP_UCODE_REG: 0X00000000
[   13.876623] iwlwifi 0000:01:00.0:           CSR_GP_DRIVER_REG: 0X00000000
[   13.883537] iwlwifi 0000:01:00.0:           CSR_UCODE_DRV_GP1: 0X00000000
[   13.890452] iwlwifi 0000:01:00.0:           CSR_UCODE_DRV_GP2: 0X00000000
[   13.897362] iwlwifi 0000:01:00.0:                 CSR_LED_REG: 0X00000020
[   13.904272] iwlwifi 0000:01:00.0:        CSR_DRAM_INT_TBL_REG: 0X88164de3
[   13.911194] iwlwifi 0000:01:00.0:        CSR_GIO_CHICKEN_BITS: 0X07800200
[   13.918106] iwlwifi 0000:01:00.0:             CSR_ANA_PLL_CFG: 0Xd55555d5
[   13.925015] iwlwifi 0000:01:00.0:      CSR_MONITOR_STATUS_REG: 0X3f7f1e10
[   13.931968] iwlwifi 0000:01:00.0:           CSR_HW_REV_WA_REG: 0X0001001a
[   13.938884] iwlwifi 0000:01:00.0:        CSR_DBG_HPET_MEM_REG: 0Xffff0000
[   13.946006] iwlwifi 0000:01:00.0: FH register values:
[   13.951210] iwlwifi 0000:01:00.0:         FH_RSCSR_CHNL0_STTS_WPTR_REG: 0X14ce8700
[   13.958910] iwlwifi 0000:01:00.0:        FH_RSCSR_CHNL0_RBDCB_BASE_REG: 0X014ce790
[   13.966628] iwlwifi 0000:01:00.0:                  FH_RSCSR_CHNL0_WPTR: 0X00000000
[   13.974363] iwlwifi 0000:01:00.0:         FH_MEM_RCSR_CHNL0_CONFIG_REG: 0X80801054
[   13.982093] iwlwifi 0000:01:00.0:          FH_MEM_RSSR_SHARED_CTRL_REG: 0X000000fc
[   13.989819] iwlwifi 0000:01:00.0:            FH_MEM_RSSR_RX_STATUS_REG: 0X07030000
[   13.997544] iwlwifi 0000:01:00.0:    FH_MEM_RSSR_RX_ENABLE_ERR_IRQ2DRV: 0X00000000
[   14.005273] iwlwifi 0000:01:00.0:                FH_TSSR_TX_STATUS_REG: 0X07ff0001
[   14.012996] iwlwifi 0000:01:00.0:                 FH_TSSR_TX_ERROR_REG: 0X00000000
[   14.020814] iwlwifi 0000:01:00.0: Start IWL Error Log Dump:
[   14.026402] iwlwifi 0000:01:00.0: Status: 0x00000000, count: 6
[   14.032260] iwlwifi 0000:01:00.0: Loaded firmware version: 22.391740.0
[   14.038811] iwlwifi 0000:01:00.0: 0x00001245 | ADVANCED_SYSASSERT          
[   14.045798] iwlwifi 0000:01:00.0: 0x000002F0 | trm_hw_status0
[   14.051554] iwlwifi 0000:01:00.0: 0x00000000 | trm_hw_status1
[... 37 lines cut ...]
[   14.259883] iwlwifi 0000:01:00.0: 0xC0081000 | umac interruptlink2
[   14.266068] iwlwifi 0000:01:00.0: 0x00000800 | umac data1
[   14.271471] iwlwifi 0000:01:00.0: 0xC0081000 | umac data2
[   14.276893] iwlwifi 0000:01:00.0: 0xDEADBEEF | umac data3
[   14.282310] iwlwifi 0000:01:00.0: 0x00000016 | umac major
[   14.287726] iwlwifi 0000:01:00.0: 0x0005FA3C | umac minor
[   14.293146] iwlwifi 0000:01:00.0: 0xC0886280 | frame pointer
[   14.298815] iwlwifi 0000:01:00.0: 0xC0886280 | stack pointer
[   14.304493] iwlwifi 0000:01:00.0: 0x0902016A | last host cmd
[   14.310167] iwlwifi 0000:01:00.0: 0x00000000 | isr status reg
[   14.320955] iwlwifi 0000:01:00.0: Failed to run INIT ucode: -5
[   14.432247] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   14.432997] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   14.551407] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   14.551918] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   14.656523] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   14.813176] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   15.087017] Bluetooth: hci0: Waiting for firmware download to complete
[   15.087325] Bluetooth: hci0: Firmware loaded in 2081000 usecs
[   15.087491] Bluetooth: hci0: Waiting for device to boot
[   15.099354] Bluetooth: hci0: Device booted in 11646 usecs
[   15.100328] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-12-16.ddc
[   15.103395] Bluetooth: hci0: Applying Intel DDC parameters completed

</code></pre>

idcrook, 2019-04-26:
After multiple failures `error -2` with other firmware, my 8265 firmware loads as yours does

```
[   10.322771] iwlwifi 0000:01:00.0: loaded firmware version 22.391740.0 op_mode iwlmvm
```

I am on the latest, just released,  `linux-firmware` package (`1.173.5`)

```shell
apt-cache show linux-firmware | head
Package: linux-firmware
Architecture: all
Version: 1.173.5
Multi-Arch: foreign
Priority: optional
Section: misc
Origin: Ubuntu
Maintainer: Ubuntu Kernel Team <kernel-team@lists.ubuntu.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 298450
```

However, am not getting the microcode error, and my wifi works as normal

```shell
dmesg | grep -e iwlwifi -e thermal_zone  -e 'wlan0: link'
...
[   12.241056] iwlwifi 0000:01:00.0: Detected Intel(R) Dual Band Wireless AC 8265, REV=0x230
[   12.243990] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.245300] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.372350] thermal thermal_zone6: failed to read out thermal zone (-5)
[   12.383107] thermal thermal_zone6: Registering thermal zone thermal_zone6 for type iwlwifi
[   12.696774] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   12.698771] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.699215] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.820128] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.820570] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.879473] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   13.111900] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   16.821093] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

jomoengineer, 2019-04-26:
Yeah, it's odd.  I powered off the board last night and this morning when powering it back up again, I don't see the error anymore.  After performing the 'apt-get update/upgrade' I just rebooted the board so I guess it required a power cycle after the install.

This is what I see from the commands you listed.
<pre><code>
$ apt-cache show linux-firmware | head
Package: linux-firmware
Architecture: all
Version: 1.173.5
Multi-Arch: foreign
Priority: optional
Section: misc
Origin: Ubuntu
Maintainer: Ubuntu Kernel Team <kernel-team@lists.ubuntu.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 298450
</code></pre>

<pre><code>
dmesg | grep -e iwlwifi -e thermal_zone  -e 'wlan0: link'
</code></pre>

All iwlwifi messages from demsg
<pre><code>
$ dmesg | grep iwl
[    9.280186] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[    9.548157] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-8265-26.ucode failed with error -2
[    9.558955] iwlwifi 0000:01:00.0: Falling back to user helper
[    9.572636] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-8265-25.ucode failed with error -2
[    9.583206] iwlwifi 0000:01:00.0: Falling back to user helper
[    9.740205] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-8265-24.ucode failed with error -2
[    9.758701] iwlwifi 0000:01:00.0: Falling back to user helper
[    9.766158] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-8265-23.ucode failed with error -2
[    9.766161] iwlwifi 0000:01:00.0: Falling back to user helper
[    9.806240] iwlwifi 0000:01:00.0: loaded firmware version 22.391740.0 op_mode iwlmvm
[    9.930444] iwlwifi 0000:01:00.0: Detected Intel(R) Dual Band Wireless AC 8265, REV=0x230
[    9.932941] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[    9.934117] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   10.053992] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'
[   10.054885] thermal thermal_zone6: Registering thermal zone thermal_zone6 for type iwlwifi
[   11.586690] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   11.587282] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   11.714574] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   11.715078] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
</code></pre>

RyanC1681, 2019-05-10:
Hi, I got similar error and Intel 8265 is not working out of box with Version: 1.173.5.
Any idea or I need to update fw?
Thanks,

jomoengineer, 2019-05-11:
Could you post the error that you are seeing with your 8265?  Preferably the commands the idcrook suggested.

RyanC1681, 2019-05-11:
Hi, here is error message as below, tried to download iwlwifi-8265-22.ucode and remove other iwlwifi-8265-*.ucode but it still didn't work. No wlan# entry in iwconfig but shown lsmod as below:  Thanks,
"iwlwifi               167043  1 iwlmvm

[   11.762356] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-8265-23.ucode failed with error -2
[   11.776791] iwlwifi 0000:01:00.0: Falling back to user helper
[   11.896769] iwlwifi 0000:01:00.0: loaded firmware version 22.361476.0 op_mode iwlmvm
[   12.016718] iwlwifi 0000:01:00.0: Detected Intel(R) Dual Band Wireless AC 8265, REV=0x230
[   12.019275] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.019938] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   12.076814] iwlwifi 0000:01:00.0: Can't parse empty family 8000 OTP/NVM sections
[   12.090132] iwlwifi 0000:01:00.0: Failed to read NVM: -61
[   12.109919] iwlwifi 0000:01:00.0: Failed to run INIT ucode: -61
[   12.123375] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled

RyanC1681, 2019-05-11:
Hi, I tried plug-in 7260 wifi that works, but I am wondering why 8265 didn't work on nano?
Thanks,

RyanC1681, 2019-05-11:
upgraded kernel to uname -sr
Linux 4.9.140-tegra
that should support 8265 that required 4.6+ with iwlwifi-8265-36.ucode
any idea?
Thanks,

RyanC1681, 2019-05-12:
Hi, 7260 works with nano image, but it diidn't work with jetbot image.
8265 didn't work on both images, even with latest kernel.
Any idea?
Thanks,

jomoengineer, 2019-05-13:
I have a 8265 in my JetBot running with the latest JetBot image and it is working fine now. 

The message in your post where the driver reports "Can't parse empty family 8000 OTP/NVM section" is concerning.  It would appear the module is either damaged, or it it is not programmed properly from the vendor.
  
<pre><code>
[ 11.896769] iwlwifi 0000:01:00.0: loaded firmware version 22.361476.0 op_mode iwlmvm
[ 12.016718] iwlwifi 0000:01:00.0: Detected Intel(R) Dual Band Wireless AC 8265, REV=0x230
[ 12.019275] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[ 12.019938] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
<b>[ 12.076814] iwlwifi 0000:01:00.0: Can't parse empty family 8000 OTP/NVM sections
[ 12.090132] iwlwifi 0000:01:00.0: Failed to read NVM: -61
[ 12.109919] iwlwifi 0000:01:00.0: Failed to run INIT ucode: -61</b>
[ 12.123375] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
</code></pre>

Looking at the Bootlin code for the Intel driver, at line 296 you can see where it is checking for the required SW and REGULATORY sections and where the error is originated from.
[https://elixir.bootlin.com/linux/latest/source/drivers/net/wireless/intel/iwlwifi/mvm/nvm.c](https://elixir.bootlin.com/linux/latest/source/drivers/net/wireless/intel/iwlwifi/mvm/nvm.c)

There seems to be two versions of the 8265 but I am not aware of a way to check this:
8265NGW
8265D2W
I'm guessing with REV=0x230 you should have the correct one. 

The Intel WiKi has more info
[https://wikidevi.com/wiki/Intel/Intel_Roadmap](https://wikidevi.com/wiki/Intel/Intel_Roadmap)

closed 2022-07-15 by jaybdub