NVIDIA-AI-IOT/jetbot #51: I encountered the situation that I could not connect to the monitor. At the beginning, I could use it, but later I could not connect to the monitor. There was no output signal

bigboss10186, 2019-04-27:
I encountered the situation that I could not connect to the monitor. At the beginning, I could use it, but later I could not connect to the monitor. There was no output signal

jaybdub, 2019-04-28:
Hi bigboss10186,

Thanks for reaching out! Are you connected to any other peripherals?  I wonder if this is a power issue.  You may also be able to reach more who can help on the [NVIDIA devtalk](https://devtalk.nvidia.com/default/board/371/jetson-nano/) forums.

Thanks!
John

bigboss10186, 2019-04-30:
When I used the 5v 2.5a charger, the monitor did not work, and I was able to use it before. The rest of the chip can be used normally

bigboss10186, 2019-05-12:
Thank you. I solved my problem. It's really the power problem

closed 2022-07-15 by jaybdub