NVIDIA-AI-IOT/jetbot #52: JupyterLab Collision Avoidance - Train Model unzip dataset.zip hangs if folder already exists

jomoengineer, 2019-04-27:
First, in order to get the train_model.ipynb file to move into the ~/Notebooks folder as described in Step 5 - Install latest software from Software Setup the following rsync command is required.  The one listed does not appear to update the existing ~/Notebooks

<pre><code>
rsync -rv   jetbot/notebooks/* /home/jetbot/Notebooks/
</code></pre>

However, under the 'Upload and extract dataset' step in Collision Avoidance - Train Model will hang if the dataset.zip step has already been performed since it is waiting for a command line response to replace the existing files.
Sorry, I can not copy from the active JupyterLab window but this is what is seen from a command prompt if run locally and is what is seen in the JupyterLab window.
<pre><code>
unzip -q dataset.zip 
replace dataset/blocked/5e351dde-68b8-11e9-bdd6-72b5f773b75d.jpg? [y]es, [n]o, [A]ll, [N]one, [r]ename:
</code></pre>

Since there is no user interaction in the Jupyter window there is no way to tell the process to continue and a Shutdown Kernel is required to end the process. 

This is screen shot of what is seen when this hangs.
[image]

Mubeeni35, 2019-07-06:
I got it to stop hanging once you delete the dataset folder and then run through the train_model notebook (make sure you just keep the dataset.zip file). It will automatically create the dataset folder and it should run optimally after that. Only way I have gotten it to work.

jomoengineer, 2019-07-08:
Yeah, that is a work around.  The other option is to just skip the step.  I'm not sure if this is a limitation of Jupyter Notebooks, but there should at least be a note or warning indicating that performing this step when a dataset folder already exists will cause the Notebook to hang and will have to be restarted.

closed 2022-07-15 by jaybdub