NVIDIA-AI-IOT/jetbot #53: Collision Avoidance - Variable "prob_blocked" always > 0.9

GPUSAIOT, 2019-04-28:
Hi Sir,

I use pre-trained model to run Collision Avoidance sample, was successful 2 days ago. Since yesterday, my Jetbot kept circles always. I check the variable "pro_blocked" by print() it, and found the value is all bigger than "0.9".
I re-flash my 128GB TF by Jetbot images several times and got same results. What's the mistake I made ?
BR

jaybdub, 2019-04-28:
Hi GPUSAIOT,

Thanks for reaching out!  Do you mind checking / sharing the following information

1.  Which camera model are you using?
2.  Is the robot in the field of the view of the camera?
3.  Did you make sure to call ``camera.observe`` after the initial execution? 
4.  Are you testing in the same environment as before? 

Best,
John

GPUSAIOT, 2019-04-28:
Hi John,
1. I use IMX219  Raspberry Pi V2 Camera with wide angle , can work normal still
2. The view is normal at the beginning, but more and more latency after running
3.  I'm not sure, please tell me how to check ?
4. This Jetbot worked two days ago, and ran normally about 4 hours. I didn't change or disassembly any parts, just had a sleep and problem came.
BR
Hawk

GPUSAIOT, 2019-04-28:
BTW, I set a "print" command here,
...................................    
    prob_blocked = float(y.flatten()[0])
>>print(prob_blocked)
    blocked_slider.value = prob_blocked
    
    if prob_blocked < 0.5:
        robot.forward(0.4)
   ........
Can see the outputs and the bar of "Blocked Rate"  are all over 0.9

jaybdub, 2019-05-02:
Hi GPUSCN,

Sorry for the late response.  Is this a network that your trained yourself? 

I think more latency is expected after adding the neural network execution in the loop, but it shouldn't grow over time.  Is that what you're experiencing?

Best,
John

ccc2876, 2019-07-18:
I would like to add additional collision detection training to my jetbot but I am not sure how to add it to the pre-trained model. Does anyone have any advice?

Thanks,
Claire

Miki-ctrl, 2021-01-13:
> BTW, I set a "print" command here,
> ...................................
> prob_blocked = float(y.flatten()[0])
> 
> > > print(prob_blocked)
> > > blocked_slider.value = prob_blocked
> 
> ```
> if prob_blocked < 0.5:
>     robot.forward(0.4)
> ```
> 
> ........
> Can see the outputs and the bar of "Blocked Rate" are all over 0.9

im having the same issue, do tell me how you solve it

closed 2022-07-15 by jaybdub