NVIDIA-AI-IOT/jetbot #58: Camera Module with low FoV?

TheNevers, 2019-05-04:
I read in the bill of materials that Jetbot uses high FoV camera. What happen if I use a low FoV camera e.g. Raspberry Pi Camera Module v2? Is there some config I need to change? Will the jetbot goes blind?

Sorry I'm very new to IoT programming/robots (Jetbot is my first "hardware" experience). I do know coding so if it's code change that's required I think I can handle. Thanks in advance.

jaybdub, 2019-05-09:
Hi TheNevers,

Thanks for reaching out!  I'm glad to hear that you've decided to give JetBot a try :)

If you use a low FoV camera, the robot will be able to see less, and may not be able to detect obstacles very close to it, or off to the side.  

We selected a high FoV camera so that the robot can see the ground beneath it, as well as a little bit above the horizon (which is good for tracking objects).

If you use a camera with different FoV, it is unlikely that the pre-trained neural networks will work well, as they are specific to the FoV and placement of the camera that we collected the dataset for.  However, we describe how to run through the data collection steps in Example 3, so you could collect your own dataset and train a model which may work!

Please let me know if you have any questions.

Thanks,
John

TheNevers, 2019-05-11:
Thanks, I’m asking because the leopard image camera has sold out 😩

closed 2019-05-15 by TheNevers