NVIDIA-AI-IOT/jetbot #59: Pre-trained model

hachpai, 2019-05-05:
Hello,

I want to use jetbot for a demo not related to IA directly, so I just want it to run autonomously while i'm perfoming some operations on video stream. Could anybody share the pre-trained model for obstacle avoidance? I'm really lacking time to produce dataset and train it by myself. 

Thank you!

hachpai, 2019-05-06:
Ok, I've found the pre-trained model on the [wiki](https://drive.google.com/open?id=1UsRax8bR3R-e-0-80KfH2zAt-IyRPtnW). A link in the notebook would be fine too :-)

jaybdub, 2019-05-06:
Hi Hachpai,

Just FYI, the dataset for that model was collected using the RPi Camera V2 in our office.  Hopefully it works for you!  If you run into any issues I'd recommend collecting your own dataset.  It doesn't take too long to get something decent :).  If you're curious, someone posted a great video on YouTube that shows the whole workflow

https://youtu.be/U3VJCSDqdG4

John

closed 2022-07-15 by jaybdub