NVIDIA-AI-IOT/jetbot #60: Can't Create gamepad controller

dennis08, 2019-05-07:
In Teleoperation, first to create gamepad controller, when I execute first cell code, there is no Controller is shown, I tried several times, always failed !!!
Gamepad is connected and the index is '1'

`
import ipywidgets.widgets as widgets

controller = widgets.Controller(index=1)  # replace with index of your controller

display(controller)
`

dennis08, 2019-05-07:
By the way.

When I execute traitlets, also get some errors.
`
from jetbot import Robot
import traitlets

robot = Robot()

left_link = traitlets.dlink((controller.axes[1], 'value'), (robot.left_motor, 'value'), transform=lambda x: -x)
right_link = traitlets.dlink((controller.axes[3], 'value'), (robot.right_motor, 'value'), transform=lambda x: -x)
`

**ERRORS:**

`
IndexError                                Traceback (most recent call last)
<ipython-input-8-ca4d2c13c16d> in <module>
      2 # the commands transfered
      3 
----> 4 left_link = traitlets.dlink((controller.axes[1], 'value'), (robot.left_motor, 'value'), transform=lambda x: -x)
      5 right_link = traitlets.dlink((controller.axes[3], 'value'), (robot.right_motor, 'value'), transform=lambda x: -x)
      6 camera_link = traitlets.dlink((camera, 'value'), (image, 'value'), transform=bgr8_to_jpeg)

IndexError: tuple index out of range
`

jaybdub, 2019-05-09:
Hi dennis08,

Typically, this means that the gamepad wasn't found (or initialized, ie by pressing a button) for the index that you selected.

1.  Check if your gamepad responds by navigating to https://html5gamepad.com/ and moving the axes around
2.  Note the index of the gamepad that responds to the axis movements, use this value in the notebook.

Best,
John

mayuanjason, 2019-05-12:
Hi dennis08,

Since there need a third party Gamepad Controller to control Jetbot, so I create a project based on Flask and HTML/JavaScript, with this project, you can only use mobile phone browser to control Jetbot.

Project: https://github.com/mayuanjason/mbot
Project Wiki: https://github.com/mayuanjason/mbot/wiki/Remote-Control

The Wiki is written in Chinese, but I think the codes should be fairly self-explanatory, you can follow the the commands listed in the Wiki in your PC(not Jetson Nano) and have a try.

Hope you enjoy it.

dennis08, 2019-05-13:
> Hi dennis08,
>
> Typically, this means that the gamepad wasn't found (or initialized, ie by pressing a button) for the index that you selected.
>
> 1. Check if your gamepad responds by navigating to [...]

Dear John,

Sure.
1. My gamepad responds by navigating to https://html5gamepad.com/ is fine, and buttons reponse is fine too.
2. And The index of my gamepad is "1". 
I try to buy another gamepad to control it.
Thnaks so much.

dennis08, 2019-05-13:
> Hi dennis08,
>
> Since there need a third party Gamepad Controller to control Jetbot, so I create a project based on Flask and HTML/JavaScript, with this project, you can only use mobile phone browser [...]

Excellent,  I will try it, thanks for your suggestion.

mayuanjason, 2019-05-16:
Hi dennis08,

I have refactoring the whole program and test it on my Jetbot, it can work now.
Just check the latest Wiki for how to run the program.
Wiki: https://github.com/mayuanjason/mbot/wiki/Remote-Control
Hope you enjoy it.

resolver101757, 2019-07-22:
i have the same error.  The game controller works on the website https://html5gamepad.com/ but not on jupyter .  ill also try the remote controll

jnie257, 2020-01-01:
I have the same problem... I have the index set correctly and have tried 3 controllers (2 no name brand controllers and an official xbox one controller) just for a sanity check. I have seen the link above but i dont want to use my phone or tablet to control this. 

For me, I have 3 scenarios... 
**1:** it works without throwing the tuple index out of range (maybe 30% of the time or less). 

**2:** it throws the tuple index out of range error and doesnt work (60% of the time or more).

**3:** it throws the tuple index out of range error and decides to work anyways. BOTH motors can be controlled using the controller despite it throwing the error at left_link.

The example code throws it even! I'm just starting out with coding... idk how to fix this yet. Hoping someone can help me out.

choconknit, 2020-02-17:
Since I only have one gamepad, I use index 0.
controller = widgets.Controller(index=0)

Morpheus2, 2020-03-03:
I too am also having the same problem.  I can see the axis moving in the widgets in the program, but no movement.

jnie257, 2020-03-03:
So if i run the example, it works! If i want to do anything other than the demo or If i copy and paste the code into a single "block", it throws a tuple index out of error when trying to use a controller for movement. This is with the correct index set and tried with 3 controllers. Can someone explain why? I mean its the EXACT same code... if it runs the block and then runs the next block, if i make just one block with all the code, it should still work. Am i missing something?

joezen777, 2020-04-29:
A couple of things. 

- unlink after successfully linking
- The remote controller USB dongle goes on the computer you're viewing the notebook on
- I had to press home twice to get two lights to show up on the game pad, and then I could link axis 5 to the right wheel, where as in any other mode only the left wheel gets hooked up to axis[1]. View the html5gamepad or starting widget display to validate you're using a valid axis.

rakshitraj, 2021-02-24:
@dennis08 The joystick is intended to be connected to the desktop machine which is remotely connected to the JetBot. I got the same error as you while the gamepad was connected to the JetBot.

alikhabazian, 2021-03-03:
Are you sure that you are connecting the gamepad to pc instead of robot?
if connected it right this would be appear 
[image]

closed 2021-07-05 by dennis08