NVIDIA-AI-IOT/jetbot #61: The base image is too large for my 64GB sd card

wjzhou, 2019-05-09:
The current base image is too large to be restored for my 64GB card. (exceed by 160MB)
I can workaround by restoring the official nano image first and copy over the APP ext4 partition

For next image could you make it a little bit smaller?

jaybdub, 2019-05-09:
Hi wjzhou,

Thanks for reaching out!  We noticed this issue before, have you tried the 63GB alternate image that we mention in the software setup?

https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup#step-1---flash-jetbot-image-onto-sd-card

Best,
John

wjzhou, 2019-05-12:
Thank you for your help. The 63GB image works.

May be be can always publish the 63GB image and use a init script to use all the space on the sd card.

closed 2019-05-12 by wjzhou