NVIDIA-AI-IOT/jetbot #62: is it possible to either use smartphone/mobile phone to teleoperate or a computer keyboard?

AndreV84, 2019-05-10:
Hi guys,
I am just wondering if it is possible to adopt the teleoperation in a way it could read from Android device instead of joystick. The issue is that I do not have any joystick and trying to understand if it can be controlled either from keyboard or additionally from a mobile phone somehow.
Thanks

salartarium, 2019-05-14:
Yes, the simplest way if you just want to use the Juptyerlab included notebooks is to just download an app that allows you to use your phone as an gamepad such as DroidPad or Ultimate Gamepad. Just search the google play store for ‘gamepad’. You can also cut and paste the basic motion notebook into the teleoperation notebook if you just want to use your phone. You can also look up how to code for ROS if you want to write your own code for this, if you want something better. You can look at the code for Nvidia’s Turtlebot for a rundown on better teleoperation, esp with a keyboard.

mayuanjason, 2019-05-16:
Hi AndreV84,

Actually, I have the same idea with you, so I wrote a program that can control Jebot by mobile phone or tablet.
Just check the Wiki for how to run the program.
Wiki: https://github.com/mayuanjason/mbot/wiki/Remote-Control
Hope you enjoy it.

closed 2022-07-15 by jaybdub