NVIDIA-AI-IOT/jetbot #65: get stuck

bigboss10186, 2019-05-12:
After many practices, I found that my jetson nsno could only run the following program once or twice. Even if I shut down kernal after using it, it would get stuck after running again. Is there any way to solve this problem.I've added swap.

jaybdub, 2019-05-16:
Hi bigboss10186,

Which model are you running?  This most frequently happens when the camera enters a bad state from shutdown.  Does it happen when you call

```bash
sudo systemctl restart nvargus-daemon
```

And then restart, and rerun the notebook?

Best,
John

bigboss10186, 2019-05-20:
I am running the object following example, after running the statement you gave me I can reinitialize the camera, I feel this is caused by the sudden shutdown

closed 2022-07-15 by jaybdub