NVIDIA-AI-IOT/jetbot #67: Issues connecting to JetBot IP address

Mubeeni35, 2019-05-20:
Hello, I am having issues connecting to the IP address of the jetbot. I'm hooking it up using an ethernet connection as I don't have the usb wifi adapter yet. Does anyone know what I may be doing wrong or how to remedy this issue? Thanks

wjzhou, 2019-05-20:
It will be easier if you can connect the a monitor and keyboard to the nano.

I used the ethernet before my wifi adapter arrived and it worked.

command to check the ip:
```
$ip addr show
```

Mubeeni35, 2019-05-27:
I am having issues connecting my laptop to the jetbot using the IP address displayed on the PiOLED display. Does anyone know what could be wrong? thanks.

hungnm2014, 2019-07-24:
I have same issues when connecting my laptop to jetbot using IP.
@Mubeeni35 : Did you connect ok? How to fix this?

Thanks

Mubeeni35, 2019-07-25:
I changed my router and everything regarding connecting to the ip was fine after that.

closed 2022-07-15 by jaybdub