NVIDIA-AI-IOT/jetbot #68: Left deviation

ddwy, 2019-05-22:
Why does my jetbot go left when going straight and backward?

aileenZhang, 2019-05-22:
Maybe you should check your power bank

jaybdub, 2019-05-23:
Hi ddwy,

Usually, due to slippage, friction and other factors throughout the system, the robot will either tend slightly to the right or left.  If the drift is extreme, could you check to see if there is anything adding resistance to one of the wheels?

Is the robot spinning in place?  If so, one of the motors may be wired backwards.

John

ddwy, 2019-05-23:
> Hi ddwy,
>
> Usually, due to slippage, friction and other factors throughout the system, the robot will either tend slightly to the right or left. If the drift is extreme, could you check to see if [...]

Hello, my situation is like this. When the car moves forward, it will move slowly to the left.

wjzhou, 2019-05-23:
I think this is unavoidable in the current open-loop control setup.
Even with the same pwm, the motors will spin slightly differently,  the car will turn slowly.

You may hold the cart, and check both motor spins as expected

ddwy, 2019-05-24:
The problem has not been solved yet.

einsteinagogo, 2019-05-25:
or because motors have different deviations!

The TT Motor described here on the PiHiut states

https://thepihut.com/products/adafruit-dc-gearbox-motor-tt-motor-200rpm-3-to-6vdc-ada3777

Note that these are very basic motors, and have no built-in encoders, speed control or positional feedback. Voltage goes in, rotation goes out! T**here will be variation from motor to motor**, so a separate feedback system is required if you need precision movement.

mine is also the same.

jaybdub, 2019-05-26:
Yep!  

Drift is unavoidable with this open-loop control scheme.  By calibrating, you can reduce the rate that the robot drifts.

Actually, the motor implementation has an ``alpha`` and ``beta`` value which may be used to linearly calibrate each motor.  The command, ``y`` sent to the motor is calculated from the input value ``x`` as

``y = alpha * x + beta``

You could adjust these parameters to linearly calibrate each motor with software.  For example,

```python
robot.left_motor.alpha = 0.3
robot.right_motor.alpha = 0.7
```

Attaching them to sliders will probably make live calibration a lot easier.  The results wont be perfect, but might help a bit.

Please let me know if you have any questions.

Best,
John

einsteinagogo, 2019-05-29:
I also found that using a better battery, for motors and Jetson Nano, deviation was less! and the robot could be controlled and go forward with less calibration.

Still needs a little tweaking, but worth checking if you battery can supply what is required.

LuisM78, 2019-11-04:
I am having the same proble. Do you know of a better robot platform for the jetson nano?
Thanks,
Luis

closed 2022-07-15 by jaybdub