NVIDIA-AI-IOT/jetbot #72: Is there any shutdown command

Andy-LZH, 2019-05-26:
I'm currently a new beginner of jetbot project, meanwhile, I'm also a beginner in Computer Science(High school student). Can some pros teach me how to shut down the system via Jupyter Lab, thanks

jaybdub, 2019-05-26:
Hi Andy-LZH,

Thanks for reaching out!  You can shutdown the JetBot from Jupyter Lab by

1.  Open a terminal using the Jupyter Lab launcher ``+``

2.  Type the following *bash* command into the terminal

    ```bash
    sudo shutdown now
    ```

FYI, if you want to reboot, you could do this ``sudo reboot now``.

Please let me know if you have any other questions!

Best,
John

Andy-LZH, 2019-05-27:
Hi John,
Thank you for your support, jetbot is a wonderful project that helped me to focus on deep learning + fun and has remarkable support from your earnest developers. Thanks a lot for your support, I have a lot of fun playing with jetbot.

Sincerely, 
Andy

closed 2019-05-27 by Andy-LZH

ArrowRaider, 2021-09-09:
System has not been booted with systemd as init system (PID 1). Can't operate.