NVIDIA-AI-IOT/jetbot #75: Fix links to the examples in Wiki

madaan, 2019-06-02:
Links to the example notebooks (like [basic_motion.ipynb](https://github.com/NVIDIA-AI-IOT/blob/master/notebooks/basic_motion/basic_motion.ipynb) are wrong in the [examples page](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/examples) in the wiki.

Seems like the links are missing the name of the repo ("jetbot"), causing the 404. For example:

https://github.com/NVIDIA-AI-IOT/blob/master/notebooks/basic_motion/basic_motion.ipynb

**_Should be:_**

https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/notebooks/basic_motion/basic_motion.ipynb

jaybdub, 2019-06-03:
Thanks for pointing this out!  Just applied this fix.

Best,
John

madaan, 2019-06-03:
Cool! While you're at it, you may want to add links to the data_collection and the live_demo notebooks (the links are currently missing).

closed 2019-06-04 by madaan