NVIDIA-AI-IOT/jetbot #76: How to use phone's voice API to control the Jetbot?

HannahZhangVW, 2019-06-04:


jaybdub, 2019-06-08:
Hi Hannah,

Do you know which phone / API you're using and how you intend to use it?  I may be able to provide more detailed guidance with that.

But as a toy example for quick guidance:  

Assume the phone's voice API is able to parse the speech and determine two commands ``stop`` and ``forward``.  We would need a way to communicate these commands to the JetBot.  One simple way, would be to create a web server that runs on the robot.  

[Flask](http://flask.pocoo.org/) is a framework for creating lightweight web applications in Python that might help here.  Following Flask's [routing rules](http://flask.pocoo.org/docs/1.0/quickstart/#routing) we could easily route paths on our server to JetBot commands

```python
# some other code ...

@app.route('/forward', methods=['POST'])
def forward():
    robot.forward()
    return 'Moving forward'

@app.route('/stop', methods=['POST'])
def stop():
    robot.stop()
    return 'Stopping'
```

If we ``POST`` to these addresses, for example ``http://<jetbot_ip_address>:<flask_port>/forward`` the robot should respond.  I think this approach is pretty portable because most programming languages support HTTP.  For example, I believe we could POST in Python as easily as this

```python
import requests
response = requests.post('http://<jetbot_ip_address>:<flask_port>/forward')
```

As a disclaimer, I haven't tested this.  But it may be a simple / portable way to communicate with the robot. 

Please let me know if you have any questions.

Thanks,
John

ccc2876, 2019-07-18:
@jaybdub 
I am working on a project with the jetbot where my team is trying to get the robot to follow different people based on voice commands. My team has purchased a few different microphones to work with and we have looked into the Hey-Jetson automatic speech recognition github but still have not found the best way to go about implementing this. Do you have any suggestions for voice recognition that would be easy to implement for our jetbot? 

Any help is appreciated!

Thanks,
Claire

closed 2022-07-15 by jaybdub