NVIDIA-AI-IOT/jetbot #77: 3D Printer Availability

Whs02, 2019-06-04:
The Monoprice 135881 printer has been unavailable for about a month.  The other alternative is out of my price range.  Can you recommend another beginner printer which will work with Jetbot files in the $300 price range?

Mubeeni35, 2019-06-04:
Why don’t you just use 3DHubs? It costs ~$50 and you receive it within 5 days. I guess I understand if you want to have your own 3d printer, then never mind

jaybdub, 2019-06-08:
Hi Whs02,

We only tested extensively with the two printers listed, but there are lots of printers out there.  Hopefully the community can provide some alternatives that they've tested with the JetBot print.  

Best,
John

Whs02, 2019-06-13:
I purchased Monoprice 38820 from Amazon.  $399 which is $80 more than the referenced printer.  It is also a Voxel and as nearly as I can tell it is identical.  I was able to print out all parts.  Probably not as good quality as experienced printer but all are good enough to use.  Note I used Flashprint for slicing and it is pretty easy for beginners.

jaybdub, 2019-06-13:
Thanks for following up Whs02.  It's valuable to know this printer worked for you.

We'll consider adding this to our list so others can benefit :)

Thanks!
John

jaybdub, 2019-06-13:
@Whs02 Just to verify.  This is the printer you used, correct?

https://www.amazon.com/Monoprice-133820-Voxel-Printer-Removable/dp/B07GV5GLLC/ref=sr_1_1_sspa?keywords=monoprice+printer&qid=1560461623&s=gateway&sr=8-1-spons&psc=1

Thanks,
John

Whs02, 2019-06-14:
Yes this is the one I used.  Most reviews were positive and the Flashprint made it possible for a total newbie to use.  I did roll the castor base and shroud 90 degrees about the x axis which seemed to make more sense for printing.  The only real pain was the inside supports for the motor holes in the chassis were difficult to clean up because of the small space.  I’ll try to do the supports differently for these next time.  Thank you for your help.

Bill

 

https://www.amazon.com/Monoprice-133820-Voxel-Printer-Removable/dp/B07GV5GLLC/ref=sr_1_1_sspa?keywords=monoprice+printer <https://www.amazon.com/Monoprice-133820-Voxel-Printer-Removable/dp/B07GV5GLLC/ref=sr_1_1_sspa?keywords=monoprice+printer&qid=1560461623&s=gateway&sr=8-1-spons&psc=1> &qid=1560461623&s=gateway&sr=8-1-spons&psc=1

closed 2022-07-15 by jaybdub