NVIDIA-AI-IOT/jetbot #79: My computer won't connect to jetbot

Mubeeni35, 2019-06-07:
When I followed to the steps to connect to the ip of the jetbot so that I can access JupyterLab, my computer will not connect to it. It says "connection timed out" or something along those lines. I can connect to it from the jetson nano itself but not when I try to from my computer. Any suggestions? thanks

jaybdub, 2019-06-08:
Hi Mubeeni35,

Thanks for reaching out!  Sounds like the Jupyter Lab server is running, but there is a connection issue between PC and Jetson.

1.  Is your JetBot on the same network as your PC?
2.  Is there any chance you have Ethernet *and* WiFi attached to the robot?

Best,
John

Mubeeni35, 2019-06-08:
Hello,
My jetbot is connected to the same network as my macbook. I am using the recommended wifi card and it is only connected to wifi. I have seen posts about changing ‘c.notebookapp.ip’ in setup_config_file.py, that has not worked for me. I have also tried a couple of other things and still no dice. I appreciate your help.

Mubeeni35, 2019-06-13:
Can anyone please help me out on this? I've spent weeks and have found 0 solutions to the problem? Thanks.

jaybdub, 2019-06-13:
Hi,

Sorry to hear this issue's persisted.  I'll try to help the best I can, mostly it sounds like a networking issue (possibly not JetBot specific).  

As a sanity test, do you mind verifying

1.  Does the IP address for the ``wlan0`` interface display on the PiOLED screen? 
2.  Are you able to ping your host machine from the Jetson Nano.  By connecting the Jetson Nano to a Monitor, Keyboard, and Mouse and calling the following from a terminal.

    ```bash
    ping <computer_ip>
    ```

I'm not a networking expert by any means, but if that succeeds, maybe the port used for Jupyter (``8888``) is blocked.

Thanks for the patience.  Hope we can get you up and running soon.  You might also reach a larger audience on our [developer forums](devtalk.nvidia.com).

Thanks,
John

Mubeeni35, 2019-06-13:
Hello John,

I appreciate your efforts to help me out so much. I just bought a new router and it works now!! Turns out the old router had settings embedded to prevent such connections and very limited options to change those settings (I’m confused, it’s 2019 and this shouldn’t even be a thing). 

Sincerely,
Muhammad

closed 2019-06-13 by Mubeeni35

jaybdub, 2019-06-13:
Glad to hear!  Hope things go smoothly now.

Please don't hesitate to reach out if you have any other questions.

Best,
John

HoangLouis, 2019-11-16:
Hi everyone, I just get the same issue here. Is there any way to solve this problem without changing the router?

Thanks,
Hoang

richard-love, 2019-11-18:
You can setup (on a Windows 10 PC at least) a DIRECT WiFi connection to the Jetbot - a shared Wifi Hotspot - note: this is not using the router, so the Jetbot will connect direct to your PC Hotspot network. Not sure if this Hotspot/Shared internet  function is also in Mac OSX, it's likely it is.

**Process**: 
1) Go to your PC Win 10 settings, in search box, type Mobile Hotspot. 
2) on the settings page shown - setup a new hotspot if one does not exist. 
3. you will need to enter a network name and password for your new hotspot
4) Connect your Jetosn Nano/Jetbot to a HDMI screen, keyboard, mouse (you only need this for 1st setup - once setup you will be able to login via Jupiter notebook.) 
5) boot the nano - then on the Ubuntu desktop you can use mouse and keyboard to go to network settings and select the hotpot network you created. 
6) login using the network password you created. Jetbot picoled screen should show show the IP assigned (or look up the IP in Ubunto network settings if you don't have the picoled screen) 
7) Nearly there, on your PC go to the browser and you can enter the IP address such as http://192.168.137.63:8888/lab?
8) done - you should now have logged into your Jetbot & it's Jupiter notbook  :-)

Hope this helps.

HoangLouis, 2019-11-24:
Thanks for sharing. I have solved my problem by this way, If someone have the same problem, try it if you don't have any solution. 
https://github.com/NVIDIA-AI-IOT/jetbot/issues/160.
I think you can allow the connection in another way, but I have already done with this way.
Hope that can help!

Best,
Hoang Louis

PatrickHo-134, 2020-03-08:
I have the same issue here and this is how I fixed it:
- Open jupyter notebook in my laptop
- After the home page displays, change the "localhost" to the IP address on piOLED
That's it. Hope it works for you!