NVIDIA-AI-IOT/jetbot #81: Streaming freezes unless jetbot is close to WiFi router

dvillevald, 2019-06-09:
I have an issue with Wifi. When I run data_collection notebook, the camera freezes occasionally and slows down so it cannot catch up (lags) unless I am very close to the router. I tried two different cameras, different Jetson Nanos, different laptops and browsers - the behavior is the same. The jetbot collects the images correctly so it seems the issue is with sending images over WiFi. I am using Intel Wireless-AC 8265 as recommended. Also, in some cases Nano fails to shut down (via sudo shutdown now) until I disconnect the camera from the board. Does someone experience a similar behavior? Any suggestions on how to fix it?

Thank you in advance!
Dmitri

jaybdub, 2019-06-12:
Hi Dmitri,

I usually experience this only when the WiFi connectivity is poor.

When collecting data, high frame-rates may not be necessary perhaps you can reduce the frame rate to limit the bandwidth.

For example, to capture at ``5fps``

```python
camera = Camera.instance(width=224, height=224, fps=5)
```

Best,
John

dvillevald, 2019-06-13:
Thank you, John, I will try to reduce the frame rate.

I wonder if you guys tested it with WiFi Dongle and, if so, how the results were. 

Do you think it might have something with quality or orientation of antennas? (e.g. they are parallel to each other.) I would love to try different orientation but it is not easy as they are hidden under the battery.

Best,
Dmitri

dvillevald, 2019-06-16:
John,

I noticed discussion in  #63 about rsyslog consuming disk space and reducing runtime memory. Per your suggestion, I disabled it

```
sudo service rsyslog stop
sudo systemctl disable rsyslog
```

and it helped to reduce latency so I am closing the issue. 
Thanks a lot!
Dmitri

closed 2019-06-16 by dvillevald