NVIDIA-AI-IOT/jetbot #84: the first example run is error

zouzhe1, 2019-06-12:
basic motion
from jetbot import Robot  ------->ok
robot =Robot()--------------error

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-b418ad9f6ab3> in <module>
----> 1 robot = Robot()

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in __init__(self, addr, freq, i2c, i2c_bus)
    229         self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
    230         self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
--> 231         self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
    232         self._pwm.setPWMFreq(self._frequency)
    233 

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in __init__(self, address, debug, i2c, i2c_bus)
     57         self.i2c = get_i2c_device(address, i2c, i2c_bus)
     58         logger.debug("Reseting PCA9685 MODE1 (without SLEEP) and MODE2")
---> 59         self.setAllPWM(0, 0)
     60         self.i2c.write8(self.__MODE2, self.__OUTDRV)
     61         self.i2c.write8(self.__MODE1, self.__ALLCALL)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setAllPWM(self, on, off)
     93     def setAllPWM(self, on, off):
     94         "Sets a all PWM channels"
---> 95         self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
     96         self.i2c.write8(self.__ALL_LED_ON_H, on >> 8)
     97         self.i2c.write8(self.__ALL_LED_OFF_L, off & 0xFF)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.3-py3.6.egg/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-0.2.3-py3.6.egg/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    254         # Send the data to the device.
    255         self._select_device(addr)
--> 256         self._device.write(data)
    257 
    258     def write_word_data(self, addr, cmd, val):

OSError: [Errno 121] Remote I/O error

Mubeeni35, 2019-06-12:
I got this error initially but it was fixed by reconnecting the wires. 
Check to see if your motor driver is connected properly to the motors. 
Try to plug in motor driver directly to the jetson nano pins without the PiOLED.

zouzhe1, 2019-06-12:
oh，its mean  the I2C  no  receive  ack  siginal？
i guessed this problem，so i connect to motor driven board，but still error，the address of I2C  jump pin  need  solder？

jaybdub, 2019-06-12:
Hi Zouzhe1,

We've noticed few common wiring errors that might cause this:

1.  The SDA / SCL wires are reversed.
2.  The 3.3V / GND wires are connected incorrectly.  Commonly it is thought that the 3.3V pin is at the corner of the motor driver board, but it's 1 pin offset from the corner.
3.  Weak solder connection

Could you double check to make sure these aren't the issue?  

Thanks,
John

zouzhe1, 2019-06-14:
i very surely  the wires  is connected OK。
i use the module of motor driver from adafruit。
[https://www.adafruit.com/product/815](url) 
because i see the chip   both are  PCA9685。
i2c address  is  default？

zouzhe1, 2019-06-14:
it is ok!i fonund my driver module A5 no connect VCC.
thank you again

closed 2019-06-15 by zouzhe1