NVIDIA-AI-IOT/jetbot #85: Connection delay error

Andy-LZH, 2019-06-13:
Initially, the jetbot ran without any issue, and the camera can provide instantaneous video with very low ping(weak delay). However after I updated the new version of the jetbot, and started to train the model within the `notebook->Collision_Avoidance` it started to become really slow, and the video's delay is more than one minute. Please help me to solve the problem, really appreciate.

dvillevald, 2019-06-13:
Did you try to move your jetbot closer to the WiFi router to see if it impacts the latency? (In my case it did - https://github.com/NVIDIA-AI-IOT/jetbot/issues/81) While the video lag is annoying, I was still able to collect the data and build the model (without using the video stream.)

Andy-LZH, 2019-06-13:
I will try it asap, in my case, it was previously totally fine, so I just want to make sure it would come back to the time when all the things is smooth. Thank you for your help.

closed 2022-07-15 by jaybdub