NVIDIA-AI-IOT/jetbot #88: In the case of jetbot, the following error occurred

ddwy, 2019-06-15:
In the case of jetbot, the following error occurred

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-18-12e166046241> in <module>
      4 robot = Robot()
      5 
----> 6 left_link = traitlets.dlink((controller.axes[1], 'value'), (robot.left_motor, 'value'), transform=lambda x: -x)
      7 right_link = traitlets.dlink((controller.axes[3], 'value'), (robot.right_motor, 'value'), transform=lambda x: -x)

IndexError: tuple index out of range

jaybdub, 2019-06-15:
Hi ddwy,

Thanks for reaching out.  This error usually occurs when the Joystick isn't initialized.

1.  Does the gamepad respond when you navigate to [https://html5gamepad.com/](https://html5gamepad.com/]?

2.  After creating the ``Controller`` class instance, did you press a button to initialize the joystick?

Best,
John

zinwalin, 2021-03-14:
> Hi ddwy,
> 
> Thanks for reaching out. This error usually occurs when the Joystick isn't initialized.
> 
> 1. Does the gamepad respond when you navigate to https://html5gamepad.com
> 2. After creating the `Controller` class instance, did you press a button to initialize the joystick?
> 
> Best,
> John

same here, most of time it shows IndexError: tuple index out of range error, I once success to get controller to make accelerating and steering.

With two steps you mentioned set up, still not working.

closed 2022-07-15 by jaybdub

villesau, 2024-07-19:
Still happens to me if the setup of the controller is in the same cell as the dlink. Even with 10s sleep in between.