NVIDIA-AI-IOT/jetbot #89: road_following example

ddwy, 2019-06-17:
Is there an instance model file for road_following????

Do you have the files needed for the road_following instance?

ddwy, 2019-06-17:
Can you provide the best_steering_model_xy.pth file?

jaybdub, 2019-06-18:
Hi ddwy,

We currently don't provide a pre-trained model for road following.  I would recommend following the ``data_collection.ipynb`` notebook to create a dataset.  

As a point of reference, we were able to train a working LEGO road follower with about ~100 images.

Currently, you'll need a game pad for the labeling portion, but we hope to soon remove this dependency to make data labeling faster, easier and cheaper.  

By the way, if you don't have LEGO pieces on hand, you could really use any visual marker as a road which may be more affordable (tape, paper, etc.)

Please let me know if you have any questions!

Best,
John

Russ76, 2019-06-19:
What I'd love to have is a model following a sidewalk. Concrete, with grass on the sides is usual.

PhotloJ, 2021-12-17:
> Hi ddwy,
>
> We currently don't provide a pre-trained model for road following. I would recommend following the `data_collection.ipynb` notebook to create a dataset.
>
> As a point of reference, we were [...]

Hi jaybdub

First of all. I'm sorry about my English. I'm not good English.

I have an issue about road following. I trained jetbot to road following with road image in paper but it's not working. Jetbot is running in non-direction. I don't know why. I trained about 200 images.