NVIDIA-AI-IOT/jetbot #90: Jetbot Direction

CLUE72, 2019-06-17:
I have a problem with the Jetbot in the Notebook Basic Motion:
if I click on the forward button the jetbot goes to the casterball and if I press backward it goes to the wheels. If I press right CCW if I press Left CW.
In Collision Avoidance - Live Demo
Jetbot goes to the casterball and crash with all. 
[image]
[image]

It seems that the engines are connected the same as in https://github.com/NVIDIA-AI-IOT/jetbot/wiki/hardware-setup
Some explanation. Thanks in advance

jaybdub, 2019-06-17:
Hi CLUE72,

Thanks for reaching out.  Sorry, as mentioned in the quote [here](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/hardware-setup#step-5---mount-motor-driver), the motor wires are reversed in the image.  Can you try flipping the red / black wires of each motor as depicted below

[image]

Please let me know if you have any questions. 

Thanks!
John

CLUE72, 2019-06-17:
Thanks now it works fine

mauroesposito, 2019-10-08:
Hiu @jaybdub ,
thank you for tip! 
Why no one update images on official documentation on hardware build steps?
m

closed 2022-07-15 by jaybdub