NVIDIA-AI-IOT/jetbot #91: Jetbot unpickling error when running train_model.ipynb

vnguyen13, 2019-06-19:
I'm running into an error when trying to run train_model.ipynb. The line that I'm at is:

`model = models.alexnet(pretrained=True)`

And I'm getting the following error:
`---------------------------------------------------------------------------
UnpicklingError                           Traceback (most recent call last)
<ipython-input-18-0a144c7ff7dc> in <module>
----> 1 model = models.alexnet(pretrained=True)

/usr/local/lib/python3.6/dist-packages/torchvision/models/alexnet.py in alexnet(pretrained, **kwargs)
     58     model = AlexNet(**kwargs)
     59     if pretrained:
---> 60         model.load_state_dict(model_zoo.load_url(model_urls['alexnet']))
     61     return model

/usr/local/lib/python3.6/dist-packages/torch/utils/model_zoo.py in load_url(url, model_dir, map_location, progress)
     65         hash_prefix = HASH_REGEX.search(filename).group(1)
     66         _download_url_to_file(url, cached_file, hash_prefix, progress=progress)
---> 67     return torch.load(cached_file, map_location=map_location)
     68 
     69 

/usr/local/lib/python3.6/dist-packages/torch/serialization.py in load(f, map_location, pickle_module)
    366         f = open(f, 'rb')
    367     try:
--> 368         return _load(f, map_location, pickle_module)
    369     finally:
    370         if new_fd:

/usr/local/lib/python3.6/dist-packages/torch/serialization.py in _load(f, map_location, pickle_module)
    530             f.seek(0)
    531 
--> 532     magic_number = pickle_module.load(f)
    533     if magic_number != MAGIC_NUMBER:
    534         raise RuntimeError("Invalid magic number; corrupt file?")

UnpicklingError: unpickling stack underflow`

I created a dataset folder after running through data_collection.ipynb. I am attempting to train directly on the Jetbot. 

Has anyone run into this issue?

jaybdub, 2019-06-19:
Hi vnguyen13,

Thanks for reaching out!  According to [this](https://stackoverflow.com/questions/23964048/python-unpickling-stack-underflow), the ``unpickling stack underflow`` error occurs when the pickled object ends abruptly.  

Perhaps the pre-trained model download was interrupted.  I believe you can find the cached models that torchvision downloads in the ``~/.torch/models`` directory.  Could you try deleting the cached model and re-calling ``model = models.alexnet(pretrained=True)``

You can delete the model by

1. Open up a terminal in Jupyter Lab  using the launcher (``+``)

2. Call the following from the terminal to change directories into ``~/.torch/models``

    ```bash
    cd ~/.torch/models
    ```
3. Call the following from the terminal to remove the cached alexnet model
    
    ```bash
    rm alexnet*.pth
    ```

Then you can try re-calling ``model = models.alexnet(pretrained=True)``.

Please let me know if this works or you run into any trouble!

Best,
John

vnguyen13, 2019-06-19:
That was it! Once I removed the cached alexnet model it ran the command and downloaded the model. Now I can move onto the next steps. 

Thanks @jaybdub !!

closed 2019-06-19 by vnguyen13