NVIDIA-AI-IOT/jetbot #93: road following error in Display Live Image code

aplouie, 2019-06-21:
When executing the (latest) code in road_following/data_collection.ipynb, the Display LIve Image code below:
----------
camera = Camera()

image_widget = widgets.Image(format='jpeg', width=224, height=224)
target_widget = widgets.Image(format='jpeg', width=224, height=224)

x_slider = widgets.FloatSlider(min=-1.0, max=1.0, step=0.001, description='x')
y_slider = widgets.FloatSlider(min=-1.0, max=1.0, step=0.001, description='y')

def display_xy(camera_image):
    image = np.copy(camera_image)
    x = x.value
    y = y.value
    x = int(x * 224 / 2 + 112)
    y = int(y * 224 / 2 + 112)
    image = cv2.circle(image, (x, y), 8, (0, 255, 0), 3)
    image = cv2.circle(image, (112, 224), 8, (0, 0,255), 3)
    image = cv2.line(image, (x,y), (112,224), (255,0,0), 3)
    jpeg_image = bgr8_to_jpeg(image)
    return jpeg_image

time.sleep(1)
traitlets.dlink((camera, 'value'), (image_widget, 'value'), transform=bgr8_to_jpeg)
traitlets.dlink((camera, 'value'), (target_widget, 'value'), transform=display_xy)

display(widgets.HBox([image_widget, target_widget]), x_slider, y_slider)
----------

throws this error:
---------
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/camera.py", line 45, in _capture_frames
    self.value = image
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 584, in __set__
    self.set(obj, value)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 573, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1205, in _notify_trait
    type='change',
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1210, in notify_change
    return self._notify_observers(change)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1247, in _notify_observers
    c(event)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 361, in _update
    self._transform(change.new))
  File "<ipython-input-2-209f56a0e6d1>", line 11, in display_xy
    x = x.value
UnboundLocalError: local variable 'x' referenced before assignment

---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-2-209f56a0e6d1> in <module>
     21 time.sleep(1)
     22 traitlets.dlink((camera, 'value'), (image_widget, 'value'), transform=bgr8_to_jpeg)
---> 23 traitlets.dlink((camera, 'value'), (target_widget, 'value'), transform=display_xy)
     24 
     25 display(widgets.HBox([image_widget, target_widget]), x_slider, y_slider)

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in __init__(self, source, target, transform)
    337         _validate_link(source, target)
    338         self.source, self.target = source, target
--> 339         self.link()
    340 
    341     def link(self):

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in link(self)
    342         try:
    343             setattr(self.target[0], self.target[1],
--> 344                     self._transform(getattr(self.source[0], self.source[1])))
    345         finally:
    346             self.source[0].observe(self._update, names=self.source[1])

<ipython-input-2-209f56a0e6d1> in display_xy(camera_image)
      9 def display_xy(camera_image):
     10     image = np.copy(camera_image)
---> 11     x = x.value
     12     y = y.value
     13     x = int(x * 224 / 2 + 112)

UnboundLocalError: local variable 'x' referenced before assignment
----------

Any ideas how to fix this?

jaybdub, 2019-06-23:
Hi aplouie,

Thanks for reaching out!  Sorry, there is a typo in the code when we refactored.  Thanks for pointing this out

It should work if you change these two lines

```python
x = x.value
y = y.value
```

to

```python
x = x_slider.value
y = y_slider.value
```

I will correct this tomorrow.

Best,
John

aplouie, 2019-06-23:
Thanks for the great tech support on a Sunday, just in time for me to fix it tonight.

closed 2022-07-15 by jaybdub