NVIDIA-AI-IOT/jetbot #95: notebooks/road_following/live_demo.ipynb

aq44634, 2019-06-25:
In notebooks/road_following/live_demo.ipynb
If `y = (0.5 - xy[1]) / 2.0` is supposed to be `y = 0.5 - xy[1] / 2.0`?

Range of y in `data_collection.ipynb` and `train_model.ipynb` is [-1, 1] with a direction which from  top of picture to the bottom.

In `y_slider = ipywidgets.FloatSlider(min=0, max=1.0, orientation='vertical', description='y')`,y_slider.value ∈[0,1] 

So i guess `y = (0.5 - xy[1]) / 2.0` is used to turn y(∈[-1, 1]) to y_slider.value (∈[0,1] )with opposite direction. 
But  `y = (0.5 - xy[1]) / 2.0` can only turn y(∈[-1, 1]) to y_slider.value (∈[-0.25,0.75] )

If my guess is true, `y = (0.5 - xy[1]) / 2.0` should be replace by `y = 0.5 - xy[1] / 2.0`

jaybdub, 2019-07-02:
Hi aq44634,

Thanks for taking the time to write this up!  I think you're correct, ``y = 0.5 - xy[1] / 2.0`` would correctly map y to the range ``[0, 1]``.  

That said, all of these approaches are approximations to the true target *angle*, which would require calibration of the camera, so this modification may effect the dynamics of the robot.

Were you able to test the road following sample with the modifications you suggest?

Thanks,
John

zsqer, 2019-12-19:
I finally find the one that have the same idea with me, the most problem of the source code is:

with np.atctan2(x, y),  y = (0.5 - xy[1]) / 2.0 can only turn y(∈[-1, 1]) to y ∈[-0.25,0.75], the point is not for the  y_slider.value i think, the key problem is the angle(=np.atctan2(x, y)) cannot get a true target value, so it's not correct. but I don't know how to fix it, Thanks aq44634, i will change the code that you fixed. And take a test to observe the difference between this two ways.

thanks

closed 2022-07-15 by jaybdub