NVIDIA-AI-IOT/jetbot #96: object following, can't import ssd_mobile_v2_coco.engine

SJTUJory, 2019-06-26:
when I run the first step,
`from jetbot import ObjectDetector
model=ObjectDetector('/home/jetbot/Notebooks/object_following/ssd_mobilenet_v2_coco.engine')`
it told me.
`---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-2-b0d090383c9d> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('/home/jetbot/Notebooks/object_following/ssd_mobilenet_v2_coco.engine')

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     26         load_plugins()
     27         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 28                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     29         self.preprocess_fn = preprocess_fn
     30 

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'`

closed 2019-06-26 by SJTUJory

santaimpersonator, 2019-07-25:
Hey there,
I just ran into this issue as well... what did you do to resolve the issue?

hungnm2014, 2019-08-30:
@SJTUJory I get same issue, could you share How to fix this
Thanks

santaimpersonator, 2019-08-30:
@hungnm2014 On my end, I narrowed it down to a version issue with TensorRT. I documented it in issue https://github.com/NVIDIA-AI-IOT/jetbot/issues/126.

HabonRoof, 2019-09-16:
Is there any solution? I have the same issue,too.

allengwinn, 2019-09-17:
I'm a n00b with this robot but it looks like ssd_mobilenet_v2_coco.engine may need to be recompiled for TensorRT 5.1.6.1. However that's done :) It complains about a versioning problem. Unfortunately the issue was apparently rectified and closed by one guy without giving much detail.

j-vara, 2019-12-11:
I have the same problem, any idea? I have read issue #126 but I don't find an answer, may be I am not undertanding it.

AK51, 2020-02-14:
I have the same problem too. is there any solution?
I use the SD card image.